modelica/ModelicaStandardLibrary #1959: Usage of pre in Modelica.Electrical.Digital erroneous

modelica-trac-importer, 2017-01-15:
**Reported by clauss on 15 Mar 2016 13:34 UTC**
P. Junglas reported errors when using basic components of Modelica.Electrical.Digital. Care for:
P. Junglas: Pitfalls using discrete event blocks in Simulink and Modelica. ASIM-Treffen STS/GMMS 2016, Treffen der ASIM/GI-Fachgruppen STS und GMMS, 10./11.März 2016, Lippstadt, pp. 90-97.

The publication is attached.

The reason of wrong behavior seems to be the usage of pre.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1959

modelica-trac-importer, 2017-01-15:
**Comment by hubertus on 15 Mar 2016 14:30 UTC**
I have not had time to analyze this fully, but it looks like it may not be possible to build a correct RSFlipFlop with Modelica 3.2.x, and that synchronous features (or an update/correction of the spec) would be needed to do so. It would also mean that a block that has been in MSL for a long time, the RSFlipFlop is not really correct. Christoph, could you add a bit more detail to your assessment of this?

modelica-trac-importer, 2017-01-15:
**Comment by hansolsson on 16 Mar 2016 13:52 UTC**
As far as I understand the reason for this is that the digital components contain pre - to break loops. If you use "long clock ticks" as the current examples in MSL it isn't a problem; and there are some benefits of this. But if you want to use instantaneous clock ticks and model a flip-flop it would require "one-sample-delay-blocks" etc - and using synchronous features becomes more convenient.

Note that there isn't really a flip-flop component in MSL; just an example model with it.

beutlich, 2017-02-01:
Point to @peterjunglas.

beutlich, 2017-03-13:
> Christoph, could you add a bit more detail to your assessment of this?

@ECClauss