modelica/ModelicaStandardLibrary #1963: Discontinuous change in angles/positions of PlanarLoopWithMove

modelica-trac-importer, 2017-01-15:
**Reported by beutlich on 18 Mar 2016 09:15 UTC**
In model _ModelicaTest.MultiBody.PlanarLoopWithMove_ there is a discontinuous change in angles/positions at (stop) time = 3 seconds since the driving _combiTimeTable_ has extrapolation set to periodic.

Is this really the desired behaviour under test?
* If yes, it would be desirable to extend the stop time to, say 5 seconds.
* If no, it would be desirable to change the extrapolation type to linear.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1963

modelica-trac-importer, 2017-01-15:
**Comment by hansolsson on 29 Mar 2016 09:54 UTC**
Replying to [ticket:1963 beutlich]:
> In model _ModelicaTest.MultiBody.PlanarLoopWithMove_ there is a discontinuous change in angles/positions at (stop) time = 3 seconds since the driving _combiTimeTable_ has extrapolation set to periodic.
> 
> Is this really the desired behaviour under test?

In that case it should also be a clearly documented since the normal assumption is that the inputs to "Move" are consistent - but a discontinuity breaks that. In this case the mechanism is determined by `r1.phi` - so a discontinuity will just lead to a dirac-impulse in the velocity and then go back to the previous behavior.

tobolar, 2018-05-29:
Well I guess the circumstances (i.e. the example documentation, identity of StopTime and table end time) indicate that the periodic extrapolation is selected by mistake. I would correct it and commit.

closed 2018-05-29 by tobolar