modelica/ModelicaStandardLibrary #1965: Modelica.Media.IdealGases.Common.Functions.h_T  with zeroDerivative for DataRecord

modelica-trac-importer, 2017-01-15:
**Reported by vitalij&#46;ruge on 25 Mar 2016 11:08 UTC**
It would be nice to have `annotation(derivative(zeroDerivative=DataRecord))`, because the guard `annotation (Inline=false) ` for inlining make the analyse difficult for modelica tools.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1965

modelica-trac-importer, 2017-01-15:
**Comment by otter on 1 Apr 2016 22:22 UTC**
This is too late for version 3.2.2, especially because the suggestion needs a careful analysis (and there is no indication that the current implementation is wrong).

modelica-trac-importer, 2017-01-15:
**Comment by hubertus on 5 Apr 2016 21:21 UTC**
This does indeed need a bit of analysis. All other similar functions have the annotation inline=true, and I don't remember and understand why this should not be inlined. If anyone reading this ticket understands or remembers the reason, please comment, it certainly does not look like this is coincidence (all enthalpy functions have inline=false). But in any case, zeroDerivative seems the correct annotation for data and could be applied to all functions in the package.

beutlich, 2017-02-07:
@hubertus65 Is there anything more to do here?

hubertus65, 2018-04-06:
The annotations in this package are definitely inconsistent and need some work. The only thing that is consistent is that derivative functions are inlined, the main functions are sometimes inlined, sometimes not. The zeroDerivative annotation should be in all functions that are not inlined (and probably that one does not matter if the code is inlined). Thomas, could you test whether inline true or false matters for SimulationX? I can check for Optimica and Dymola. I think there was a reason to do it a specific way for Dymola. I will add the zeroDerivative to functions with inline=false, that is certainly correct.

hubertus65, 2018-04-06:
It looks like the restriction that "inline=true" did not work is a long gone restriction from some tool. Inlining works with both Optimica and Dymola. Inlining should, in general, be faster, so I am tempted to change all annotations to inline=true, in which case the derivative annotation is not needed. Now it would be nice to have automatic regression testing with all tools, because I could break things with a tool that I did not test. Comments?

hubertus65, 2018-04-06:
Caveat: the manually written derivatives could be faster, it is a nicely written horner schema.

carlj-w, 2020-02-26:
Coming back to the specific function _Modelica.Media.IdealGases.Common.Functions.h_T_: In the same package there is _Modelica.Media.IdealGases.Common.Functions.h_T_der_, which seems to be the derivative of the function, but _h_T_ does not have a derivative annotation pointing to _h_T_der_. This seems strange to me.

One can notice the same for other functions in the same package, e.g. _h_Tlow_ and _h_Tlow_der_.

thorade, 2020-02-26:
Related: https://github.com/modelica/ModelicaStandardLibrary/issues/1909
Especially, follow the links at the bottom to other issues where #1909 was mentioned.

hubertus65, 2023-03-08:
This is a duplicate of an issue that has been closed quite a while ago, see resolution to #1909

hubertus65, 2023-03-08:
Has been closed earlier

closed 2023-03-08 by hubertus65