modelica/ModelicaStandardLibrary #1966: Evaluate parameter phase index in Modelica.Electrical.MultiPhase.Basic.PlugToPin_n

modelica-trac-importer, 2017-01-15:
**Reported by vitalij&#46;ruge on 26 Mar 2016 19:40 UTC**
replace
```modelica
parameter Integer k(final min=1, final max=m, start = 1) "Phase index";
```
with
```modelica
parameter Integer k(final min=1, final max=m, start = 1) "Phase index" annotation(Evaluate=true);
```
would helpful for modelica-tools.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1966

modelica-trac-importer, 2017-01-15:
**Comment by hansolsson on 29 Mar 2016 09:35 UTC**
I am not sure that this annotation is needed, and if we add it here we would have to add it in many other models.

The model contains:
` pin_n.v = plug_n.pin[k].v;`
if a tool can handle this variable indexing I don't see that it will be that inefficient (in general). Obviously variable indexing (especially for array of components) has a certain cost and tools may avoid it, but I don't see that the cost is particularly high for this case.

I even tested to make Modelica.Electrical.Machines.Examples.Transformers.AsymmetricalLoad work for non-evaluated 'k' - as far as I can see it requires replacing the initial equation by:
`  transformer.i2[plugToPin_n.k] = 0;` or preferably `load.i=0`, but I don't see that it leads to larger (or additional) system of equations during simulation.

modelica-trac-importer, 2017-01-15:
**Comment by vitalij&#46;ruge on 29 Mar 2016 22:21 UTC**
Replying to [comment:1 hansolsson]:
> I am not sure that this annotation is needed, and if we add it here we would have to add it in many other models.
>
> The model contains:
> ` pin_n.v = plug_n.pin[k].v;`
> if a tool can handle this [...]

How about the relation `j == k` inside
```modelica
  for j in 1:m loop
    plug_n.pin[j].i = if j == k then -pin_n.i else 0;
  end for;
```

I will be happy if it is possible to simplify(can easy inderduce by evalute `k`) to:
```modelica
  for j in 1:k-1 loop
    plug_n.pin[j].i = 0;
  end for;
  plug_n.pin[k].i = -pin_n.i;
  for j in k+1:m loop
    plug_n.pin[j].i = 0;
  end for;
```
with (many)`plug_n.pin[j].i = 0;`, which can replace by modelica tool.

henrikt-ma, 2018-11-21:
Hans, do you mean that the `for`­ equation
```
  for j in 1:m loop
    plug_n.pin[j].i = if j == k then -pin_n.i else 0;
  end for;
```
is expected to be treated as a single multi-dimensional equation where a lot of intelligence is applied to figure out that `pin_n.i` is actually always present in the equation's `m`-dimensional residual, where the number `m` isn't even known during translation?

I would find it much more natural if the intention was to handle `m` and `k` as constants (and they should be declared as such), so that the `for` equation can be unrolled with `pin_n.i` being present in exactly one of the loop iterations.

HansOlsson, 2018-11-23:
> I would find it much more natural if the intention was to handle m and k as constants (and they should be declared as such), so that the for equation can be unrolled with pin_n.i being present in exactly one of the loop iterations.

My point was that it is possible to handle that case without evaluating k - and it was unclear why we need this. The statement "helpful for Modelica tools" wasn't a clear problem description. 

In particular there are two main considerations:

- Are we doing it to improve performance and not just to be helpful for the Modelica tools? (Similarly as evaluating rotational axis in MBS.) If so we should have some models showing that unevaluated 'k' is less efficient. Above it seems I didn't find any difference, but now when I now retest the model above it seems that non-evaluated 'k' leads to larger non-linear system of equations.
- There seems to be other similar cases of parameters that are evaluated in practice where an annotation could be clearer.

henrikt-ma, 2018-11-23:
The {{Evaluate = true}} annotation is definitely not more clear than making it a constant.

HansOlsson, 2018-11-23:
It is intended to be changed - and as indicated above it could even be changed after translation, so it should be a parameter - not a constant.

henrikt-ma, 2018-11-26:
I can see that you indicate that `k` _could_ be handled in such a way that it can be changed after translation.  My points are:
- It seems relatively difficult for tools to verify that `pin_n.i` is actually present in the equations, so if we don't want to make assumptions about causality when defining `PlugToPin_n`, some tools may have to resort to guessing that `pin_n.i` can be solved for, depending on causality in the model.
- As you've noticed yourself, the models may be solved more efficiently when `k` is known at translation time.
- Marking something with `annotation(Evaluate = true)` goes against the excellent rule that annotations should not affect simulation behavior.  Making `k` a constant is a much better way to show both users and tools that `k` doesn't change after translation.  That `k` is intended to be changed should be indicated using its documentation string and the `Dialog` annotation.

HansOlsson, 2018-11-26:
Adding `annotation(Evaluate=true)` to k can be done, but I don't see that we can change more since:

The value of 'k' is bounded by 'm', and in most (all?) tools you cannot change 'm' after translation (since 'm' gives the size for an array of components). 
In general if 'k' is bounded by 'm' then 'k' should be at least as variable as 'm'. However, 'm' is just a parameter - it doesn't even have `annotation(Evaluate=true)` (it is sort of implicit from the fact that it gives size for an array of components).

henrikt-ma, 2018-11-26:
Yes, and for this reason both `k` and `m` should have constant variability.

PR #3119 (2020-02-02): Feature/phase constants
This PR is limited to `Modelica.Electrical` and `Modelica.Magnetic`.  It does for these sub-libraries what #3113 does for `Modelica.Mechanics`, namely change some components declared `parameter` to be `constant` instead in order to make clear to both users and tools that they can and are expected to be evaluated during translation.

There is ongoing discussion in #3113 whether the right way to do this would be with `Evaluate=true` instead of `constant`, and the two PRs should be settled the same way.

Note connection to #1966, which is fixed by the first commit.

Fix #1 [... cut]
commit 6bd7d95a (2020-02-02): Set Evaluate=true on "Phase index" and friends

This is a constant-free replacement for 800c702 'Change variability of "Phase index" and friends from parameter to constant'.

Let's hope that one doesn't forget to set Evaluate=true when adding new components in the future... (With the use of constant, the Modelica tool would let you know when you accidentally try to introduce a component with a parameter instead of a constant.)

Fixes #1966
closed 2020-02-02 by beutlich