modelica/ModelicaStandardLibrary #1976: createDirectory fails on existing directory

modelica-trac-importer, 2017-01-15:
**Reported by beutlich on 4 Apr 2016 13:39 UTC**
_Modelica.Utilities.Files.createDirectory_ fails if directory exists and _directoryName_ has a trailing slash. The error message is:
```
Error
In Modelica.Utilities.Internal.FileSystem.mkdir: Not possible to create new directory
"C:/temp":
File exists
```

To reproduce on Windows, create a directory C:\temp\ and run the following two models

```mo
model A "Model succeeds"
  parameter String d1 = "C:/temp";
  initial algorithm
    Modelica.Utilities.Files.createDirectory(dir);
end A;
```

```mo
model B "Model fails"
  parameter String dir = "C:/temp/";
  initial algorithm
    Modelica.Utilities.Files.createDirectory(dir);
end B;
```

Usually, directory names are distinguished from file names by a trailing slash. Thus, I'd have expected that createDirectory also works on names with a trailing slash. At least it should be documented.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1976

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 5 Apr 2016 08:33 UTC**
This is the customer's workaround (just in case anyone else stumbles upon the same issue).
```mo
if Strings.length(directoryName) == Strings.findLast(directoryName, "/") then
  createDirectory(Strings.substring(directoryName, 1, Strings.length(directoryName) - 1));
elseif Strings.length(directoryName) == Strings.findLast(directoryName, "\\") then
  createDirectory(Strings.substring(directoryName, 1, Strings.length(directoryName) - 1));
else
  createDirectory(directoryName);
end if;
```

modelica-trac-importer, 2017-01-15:
**Modified by beutlich on 5 Apr 2016 08:34 UTC**

beutlich, 2017-01-31:
@MartinOtter Unfortunetaly, the changes added by ed6b9a08db42be43341319d2cdf014988a606ba0 considered the case of `a:\`, but forgot to deal with `a:\test1`, though the comment looked promising:
```c
/* For some reason _stat requires a:\ and a:\test1 and fail on a: and a:\test1\ */
/* It could be handled in the Modelica-code, but seem better to have here */
```
It is now fixed in branches master and maint/3.2.2.

closed 2017-01-31 by beutlich