modelica/ModelicaStandardLibrary #1977: Routines for bug fixing in MSL

modelica-trac-importer, 2017-01-15:
**Reported by hubertus on 4 Apr 2016 13:42 UTC**
The late discovery of a relatively serious backwards compatibility issue during the release of MSL 3.2.2 made it obvious that it woudl be useful to have a faster and more responsive process to fix bugs than the one that we have for a full release of the MSL. This ticket has the purpose to collect input from members of MAP-Lib and vendors to come up with a quick and responsive process for fixing bugs in released versions of MSL. 

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1977

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 4 Apr 2016 14:18 UTC**
Did not get a mail once this ticket was created.

modelica-trac-importer, 2017-01-15:
**Comment by hansolsson on 7 Apr 2016 09:56 UTC**
Since many didn't see the ticket I reply in full.

Replying to [ticket:1977 hubertus]:
> The late discovery of a relatively serious backwards compatibility issue during the release of MSL 3.2.2 made it obvious that it woudl be useful to have a faster and more responsive process to fix [...]

I agree that we need to improve this.

To me there are two issues:
1. Actually having bug-fix releases of MSL with no unnecessary changes that can be released in an efficient way.
2. During testing of a new release (with new functionality) allow bug-fixes (not just documentation changes) without unnecessary delays of the release process.

I believe both of these are good, although one of them might if necessary suffice (as I understand it is especially difficult to get people interested in a pure bug-fix release).

It also indicates that adding strict and time-consuming processes, as we have done for the MSL release, might look better on paper than in reality. 

Having simpler rules for releases such as MSL 3.2.2 - and keep the previous ones for releases such as MSL 3.2.0 (and 3.3.0) - is also a possibility (or just keep previous rules for MSL 3.0.0 and MSL 4.0.0?). But even if new major releases are even rarer, I fear that we would still face the same problem in that case - and I am not convinced that having different processes is the best solution.

modelica-trac-importer, 2017-01-15:
**Modified by otter on 5 Aug 2016 05:59 UTC**

beutlich, 2018-03-21:
@hubertus65 @HansOlsson Can this be closed with the new MAP-LIB project rules (almost) in place?

HansOlsson, 2018-03-22:
> Can this be closed with the new MAP-LIB project rules in place?

Yes, as far as I can see the new project rules give guidance without enforcing strict rules.

closed 2018-03-22 by beutlich