modelica/ModelicaStandardLibrary #1984: Comment in Spice3.mo

modelica-trac-importer, 2017-01-15:
**Reported by eshmoylova on 12 Apr 2016 18:35 UTC**
In MSL Electrical/Spice3.mo there is a comment for model Modelica.Electrical.Spice3.Examples.Spice3BenchmarkFourBitBinaryAdder that I find rather passive aggressive and not very professional. It says:
```
// *** (FOR THOSE WITH MONEY (AND MEMORY) TO BURN)
```

The comment is present in the code, and also appears in the HTML documentation. As far as I can see it is still present in the latest version. Do you find this comment is appropriate, necessary, and professional? Should it be kept as is?

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1984

modelica-trac-importer, 2017-01-15:
**Comment by hansolsson on 13 Apr 2016 08:38 UTC**
The string is seen twice.

As far as I understand it was part of the original spice3 netlist (see e.g. https://newton.ex.ac.uk/teaching/CDHW/Electronics2/userguide/secA.html#A.4 ) and in the that part of the documentation we should either keep the entire spice3 netlist intact or - just replace it with a reference - filtering out specific comments look odd.

However, we should remove the comment in the Modelica code - not only because it looks unprofessional, but also it makes no sense and causes more question than it answers - assumedly some part of the model is expensive - but we don't know which one; and it is possible that it refers to a transistor that I cannot find in the Modelica-model.

modelica-trac-importer, 2017-01-15:
**Modified by dietmarw on 13 Apr 2016 08:57 UTC**

beutlich, 2017-02-07:
@kristinmajetta Do you agree to keep the comment in the Documentation and remove it from the (generated) Modelica code?

Pointing to @eshmoylova.

beutlich, 2017-10-04:
OK, closing then.

closed 2017-10-04 by beutlich