modelica/ModelicaStandardLibrary #1991: Lots of new HTML errors in latest MSL release

modelica-trac-importer, 2017-01-15:
**Reported by dietmarw on 28 Apr 2016 11:58 UTC**
The latest MSL release 3.2.2 contains again lots of HTML errors that need to be fixed. 

The main reason for the increase are the newly added libraries. 

The OpenModelica projects runs the HTML documentation generator on a regular basis which also produces the error log for all libraries (including MSL). 

So **each** library officer should have a look at 
https://build.openmodelica.org/Documentation/tidy.filtered and simply search for "`Modelica.`" in order to jump to the part where the MSL related errors are reported

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1991

modelica-trac-importer, 2017-01-15:
**Comment by otter on 28 Apr 2016 13:09 UTC**
I wonder whether OpenModelica could improve its output so that the line numbers of the warnings/errors have a relationship to the info-layer. Otherwise, it is hard to figure out the issue.

Most of the warnings look very similar and I tried to understand the warning messages for Modelica.Blocks.Examples.NoiseExamples.AutomaticSeed

```
[Checking file Modelica.Blocks.Examples.NoiseExamples.AutomaticSeed.html]:
line 25 column 14 - Warning: inserting implicit <p>
line 35 column 14 - Warning: inserting implicit <p>
line 60 column 1 - Warning: inserting implicit <p>
line 23 column 1 - Warning: trimming empty <p>
line 25 column 14 - Warning: trimming empty <p>
line 33 column 1 - Warning: trimming empty <p>
line 35 column 14 - Warning: trimming empty <p>
line 42 column 1 - Warning: trimming empty <p>
line 60 column 1 - Warning: trimming empty <p>
```

The info-layer of this model is:

```
<html>
<p>
This example demonstrates manual and automatic seed selection of
<a href="Modelica.Blocks.Noise.UniformNoise">UniformNoise</a> blocks, as well
as starting the noise at startTime = 0.5 s with an output value of y = -1 before this
time. All noise blocks in this example generate uniform noise in the
band y_min=-1 .. y_max=3 with samplePeriod = 0.01 s.
</p>

<p>
The blocks automaticSeed1, automaticSeed2, automaticSeed3 use the default
option to automatically initialize the pseudo random number generators
of the respective block. As a result, different noise is generated, see next
diagram:
</p>

<p><blockquote>
<img src="modelica://Modelica/Resources/Images/Blocks/NoiseExamples/AutomaticSeed1.png">
</blockquote></p>

<p>
The blocks manualSeed1, manualSeed2, manualSeed3 use manual selection of the local seed
(useAutomaticLocalSeed = false). They use a fixedLocalSeed of 1, 2, and 3 respectively.
Again, different noise is generated, see next diagram:
</p>

<p><blockquote>
<img src="modelica://Modelica/Resources/Images/Blocks/NoiseExamples/AutomaticSeed2.png">
</blockquote></p>

<p>
Try to set fixedLocalSeed = 1 in block manualSeed2. As a result, the blocks manualSeed1 and
manualSeed2 will produce exactly the same noise.
</p>
</html>
```

The issue was not obvious for me.

I pasted the above html code in a text file and tried tidy in Firefox. Now, nice marking and corresponding warnings are visible. The issue seems to be that <blockquote> is not allowed to be nested in <p></p>. This seems to be a systematic (uncritical) error in MSL.

Changed with a text editor automatically the above pattern to the correct one in 0f652b8509ddf82f9cfb5b85f42259133e55ed8a (57 occurrences).

modelica-trac-importer, 2017-01-15:
**Comment by dietmarw on 28 Apr 2016 14:03 UTC**
In this case you have to blame tidy as it is not OpenModelica itself giving the error messages but simply displaying what tidy reports. In practice this means that the report of the log can only give you an indicator which then can be used to investigate further. The line numbers are normally referred to the lines of HTML and match quite good the line numbers within the documentation annotation.

So I think it is not fair blaming OpenModelica for not doing a good enough job or even tidy to not being verbose enough. At least OpenModelica offers this service whereas other tools don't report **anything** and even allow the generation ot illegal HTML constructs. So if you want to blame a tool, blame the one which was used to create the erroneous HTML in the first place ;-).

modelica-trac-importer, 2017-01-15:
**Comment by dietmarw on 28 Apr 2016 14:30 UTC**
I forgot to point also to the log file that warns about empty link targets. For example when one links to a non existing class in the documentation. This file can be found in:

https://build.openmodelica.org/Documentation/FindFiles.log

Again search for lines beginning with `Modelica.` to find MSL relevant mistakes.

<sub>(I've also updated the description of this ticket with that log file link)</sub>

beutlich, 2017-02-09:
@sjoelund The one missing file `Phi(ref)=Phi,re jPhi,im.png` from `Modelica.Magnetic.QuasiStatic.FundamentalWave.UsersGuide.Concept.html` looks like a false positive since the link is correct. Could it be due to the commas in the file name?

beutlich, 2017-02-12:
@sjoelund I fixed most of the issues. Can you please let OMC regenerate the tidy.filtered and FindFiles.log reports. Thank you.

sjoelund, 2017-02-13:
I guess there will be a build on Wednesday evening. There was a problem with too many html-files being generated for bash to handle, so I need a new OMC build and that's done in tomorrow's nightly which means 2 days from now the documentation job should process everything automatically...

beutlich, 2017-02-14:
@sjoelund The current sorting across libraries in FindFiles.log is not that natural.
```
Modelica.Blocks.
ModelicaCompliance.
Modelica_DeviceDrivers.
ModelicaDEVS.
Modelica.Electrical.
Modelica_LinearSystems2.
Modelica.Magnetic.
Modelica.Math.
Modelica_Noise.
Modelica.UsersGuide.
```
Do you think you can have it sorted by libraries?

tbeu, 2017-02-16:
@sjoelund Thursday morning at 8 o'clock, tidy.filtered and FindFiles.log were not updated. Can you please check. Thanks a lot!

beutlich, 2017-02-17:
@sjoelund Any idea why tidy.filtered and FindFiles.log are still not up-to-date?

sjoelund, 2017-02-17:
https://test.openmodelica.org/hudson/job/OpenModelica_DOC/1440/console :( I guess I forgot to account for html-files in resources directories...

sjoelund, 2017-02-23:
@tbeu https://test.openmodelica.org/hudson/job/OpenModelica_DOC/ says it completed successfully now.

tbeu, 2017-02-23:
Thanks for confirmation.

closed 2017-02-23 by tbeu