modelica/ModelicaStandardLibrary #1995: Incorrect homotopy in Modelica.Fluid.Machines.BaseClasses.PartialPump

modelica-trac-importer, 2017-01-15:
**Reported by hansolsson on 10 May 2016 08:00 UTC**
The equation (actually two of them):
```
head = homotopy((N/N_nominal)^2*flowCharacteristic(V_flow_single* N_nominal/N), 
N/N_nominal*(flowCharacteristic(0)+delta_head_init*V_flow_single));
```
is not unit-correct; in the simplified version delta_head_init\*V_flow_single has the correct unit m<sup>4</sup>/s but flowCharacteristic(0) has unit m.

I assume the idea was to use a linearization such as `f(x)=f(0)+(f(x0)-f(0))*x/x0` to compute this - but the `x0` part is missing.

Based on that it should be:
```
head = homotopy((N/N_nominal)^2*flowCharacteristic(V_flow_single* N_nominal/N), 
N/N_nominal*(flowCharacteristic(0)*(N/N_nominal)+delta_head_init*V_flow_single/V_flow_single_init));
```
Which is unit-correct, but so is the simpler variant:
```
head = homotopy((N/N_nominal)^2*flowCharacteristic(V_flow_single* N_nominal/N), 
N/N_nominal*(flowCharacteristic(0)+delta_head_init*V_flow_single/V_flow_single_init));
```
However, in practice both would cause errors since V_flow_single_init is as default 0 - which would lead to division by zero (actually zero divided by zero since delta_head_init is also zero in those cases).

Thus the actual corrections is more complex.

Note that as long as the homotopy-solver isn't activated this will not cause any problems.

The issues has been present at least since MSL 3.2.1; maybe earlier. It isn't detected by Dymola 2016 FD01 and earlier due to an unrelated issue.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1995

modelica-trac-importer, 2017-01-15:
**Comment by hansolsson on 10 May 2016 08:23 UTC**
Something that was indirectly stated before (but should be emphasized is that): as long as system.m_flow_start is zero (the default) the unit-incorrect part adds zero and can thus be ignored.

Correction: The first sentence of the ticket should say: 

The equation (actually two of them):
```
head = homotopy((N/N_nominal)^2*flowCharacteristic(V_flow_single* N_nominal/N), 
N/N_nominal*(flowCharacteristic(0)+delta_head_init*V_flow_single));
```
is not unit-correct; in the simplified version delta_head_init\*V_flow_single has the **in**correct unit m<sup>4</sup>/s but flowCharacteristic(0) has the **correct** unit m.

modelica-trac-importer, 2017-01-15:
**Modified by fcasella on 11 Nov 2016 15:33 UTC**

modelica-trac-importer, 2017-01-15:
**Modified by fcasella on 11 Nov 2016 15:34 UTC**

HansOlsson, 2017-06-15:
It seems this one is only partially corrected and ModelicaTest.Fluid.TestComponents.Machines.TestControlledPump  thus fails with division by zero.

PR #2528 (2018-05-28): Avoid division by zero in default homotopy-code for Modelica.Fluid.Machines.BaseClasses.PartialPump
The change dc4f6d3e2a0f27b72f7146e72b9cdaaddc08d3a4 for #1995 / #2101 causes a division by zero if homotopy is activated in default configuration, since system.m_flow_start is as default zero. This change avoid the division by zero by just skipping the extra part - restoring previous behavior.

Closes #1995
closed 2018-05-28 by casella