modelica/ModelicaStandardLibrary #1999: Connection to disabled frame_resolve

modelica-trac-importer, 2017-01-15:
**Reported by hansolsson on 23 May 2016 08:56 UTC**
Modelica.Mechanics.MultiBody.Examples.Rotational3DEffects.ActuatedDrive 
and 
Modelica.Mechanics.MultiBody.Examples.Rotational3DEffects.MovingActuatedDrive
have connections to disabled frame_resolve connetors (for torque).

I don't see any explanation for this, and just believe it is a bit odd.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1999

beutlich, 2017-02-07:
@MartinOtter I wanted to delete both invalid connect statements but wondered why they have set `LinePattern.Dot` for the connection lines set by purpose.

closed 2018-01-10 by tobolar

HansOlsson, 2018-01-10:
> @MartinOtter I wanted to delete both invalid connect statements but wondered why they have set LinePattern.Dot for the connection lines set by purpose.

@beutlich The pattern likely means nothing and the explanation is as follows:
If you draw the connection line **from** the resolve-frame in Dymola (assuming it is enabled) the connection-line will "inherit" the LinePattern.Dot from that connector (same logic as for inheriting color).

commit a56e5ba8 (2018-01-10): MultiBody: Removed obsolete connections in examples concerning rotational 3D effects

Resolves #1999