modelica/ModelicaStandardLibrary #2000: Modelica.Fluid.Examples.DrumBoiler: incorrect unit on output p_S

modelica-trac-importer, 2017-01-15:
**Reported by dag on 24 May 2016 11:25 UTC**
The top-level output `p_S` in Modelica.Fluid.Examples.DrumBoiler.DrumBoiler has incorrect unit. The sensor value (in Pa) is converted to Bar using a gain-block. However, the unit is not adjusted so the unit remains Pa although the value is scaled.

The minimum update would be to add the appropriate unit on the output of `Pa2bar `with a modifier

```
y(unit="bar")
```
I think an even better change would be to delete the components `Pa2bar `and `K2degC` altogether, because the outputs from the corresponding sensors already have the displayUnits we want.

 

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/2000

modelica-trac-importer, 2017-01-15:
**Comment by rfranke on 18 Jun 2016 10:06 UTC**
Good point. It is not an option to just remove the conversion blocks, because the global outputs are intended for model export, where the actual values must be scaled appropriately, not just the display. 

Think of the use case that one wants to build a controller around the DrumBoiler model, e.g. using Modelica.Blocks. It is very unlikely that the controller signals would be scaled in Pascal or Kelvin anywhere in the world.

The dedicated Celsius.FromKelvin K2degC is working correctly. We might add dedicted `toBar` and `fromMW` blocks to MSL. Where?

Alternatively: Modelica sensors should support unit conversion of the signal values that can be modified for the application at hand. An imperial application would most likely want to use psi and degF.

modelica-trac-importer, 2017-01-15:
**Modified by rfranke on 18 Jun 2016 10:15 UTC**

beutlich, 2017-02-07:
@rfranke Would Modelica.SIunits.Conversions.to_bar be an option?

DagBruck, 2017-03-03:
@beutlich The minimum change that fixes the problem is to add the `y(unit="bar")` modifier.

henrikt-ma, 2017-03-03:
Hej!

Tack för senast.

Märkte du att Hans placerade dig i gruppen som skulle styra upp tolkningen av arrowSize?
https://trac.modelica.org/Modelica/ticket/1894 <https://trac.modelica.org/Modelica/ticket/1894>

Jag har gjort ett konkret förslag som jag hoppas vi ska kunna göra klart och få med i 3.4.  Jag räknar med att Stefan Vorkoetter ger sina synpunkter så fort han är på benen hemma i Kanada igen.

Henrik

beutlich, 2017-04-12:
Resolved in master by #2197.

closed 2017-04-12 by beutlich