modelica/ModelicaStandardLibrary #2001: ModelicaExternalC not compatible with hdf5 1.10

modelica-trac-importer, 2017-01-15:
**Reported by perost on 31 May 2016 14:05 UTC**
ModelicaExternalC doesn't build with the latest release of the hdf5 library, since the arguments to the macro H5Rdereference has changed¹. When trying to build ModelicaExternalC (using autotools) I get the following errors with hdf5 1.10:
```
../../C-Sources/ModelicaMatIO.c:18392:69: error: too few arguments to function call, expected 4, have 3
                ref_id = H5Rdereference(dset_id,H5R_OBJECT,ref_ids+i);
                         ~~~~~~~~~~~~~~                             ^
/usr/include/H5Rpublic.h:63:8: note: 'H5Rdereference2' declared here
H5_DLL hid_t H5Rdereference2(hid_t obj_id, hid_t oapl_id, H5R_type_t ref_type, const void *ref);
       ^
<... several more errors saying the same thing ...>
```

¹ https://www.hdfgroup.org/HDF5/doc/RM/RM_H5R.html#Reference-Dereference

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/2001

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 31 May 2016 14:29 UTC**
This is a duplicate of https://github.com/tbeu/matio/issues/37. It will be fixed by the provided patch (upstream) once matio v1.5.7 is released such that I can merge it to the external C code.

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 21 Jun 2016 19:58 UTC**
Resolved by ab69b8c1c2de2b7406d2a79f9a850c98020aca90. Sorry for the delay.

modelica-trac-importer, 2017-01-15:
**Changelog modified by beutlich on 21 Jun 2016 19:58 UTC**
Updated matio library to v1.5.7 (ModelicaMatIO.[ch])

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 3 Jul 2016 20:10 UTC**
Updated ModelicaMatIO.[ch] to v1.5.8 by 4ae6fed3f1420af0d77c463231d22e23c099da74.

modelica-trac-importer, 2017-01-15:
**Changelog modified by beutlich on 3 Jul 2016 20:10 UTC**
Updated matio library to v1.5.8 (ModelicaMatIO.[ch])

closed 2017-01-15 by modelica-trac-importer

adangert, 2018-10-11:
I got this error today.

beutlich, 2018-10-11:
> 
> 
> I got this error today.

Already fixed.