modelica/ModelicaStandardLibrary #2003: Use of undeclared variables in Dialog.enable, Modelica.Electrical

modelica-trac-importer, 2017-01-15:
**Reported by hansolsson on 3 Jun 2016 11:43 UTC**
Undeclared variable: tableOnFile
in the Dialog.enable attribute of fileName declared in class Modelica.Electrical.Digital.Memories.DLATRAM

Undeclared variable: tableOnFile
in the Dialog.enable attribute of fileName declared in class Modelica.Electrical.Digital.Memories.DLATROM
\--
Technically the following is not in Electrical.Digital - but we lack a "Component" for it:

Undeclared variable: useFilter
in the Dialog.enable attribute of pulse2.twoPulse.filter[1].yStart declared in class Modelica.Electrical.PowerConverters.ACDC.Control.Filter
\--
(There are also at least one other case in another MSL-part; will open separate tickets. Thought this was better than re-opening #1632 )

I don't know if it just should be removed, or there is some deeper meaning to it.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/2003

modelica-trac-importer, 2017-01-15:
**Comment by hansolsson on 3 Jun 2016 12:14 UTC**
Together with #2004 these are all the cases in MSL.

modelica-trac-importer, 2017-01-15:
**Comment by majetta on 8 Jun 2016 04:38 UTC**
I think we should set the enable attribute to true since I suppose this was meant by setting enable=tableOnFile in the first time. Maybe in an older version of the Modelica spec enable=tableOnFile was valid?

modelica-trac-importer, 2017-01-15:
**Comment by hansolsson on 20 Jun 2016 07:40 UTC**
Replying to [comment:2 majetta]:
> I think we should set the enable attribute to true since I suppose this was meant by setting enable=tableOnFile in the first time. Maybe in an older version of the Modelica spec enable=tableOnFile was valid?

(Thought I had added this.)

Not in the spec - but it could be that:
* fileName with annotation was copied from some table-model; but "tableOnFile" wasn't copied.
* Originally the model had "tableOnFile" as well, but it was removed - perhaps it wasn't implemented, or the implementation seemed excessive.

But the reason doesn't matter so much, if we want to support having the table in the model we can implement that - otherwise remove the parameter. Both ideas are fine with me.

modelica-trac-importer, 2017-01-15:
**Comment by hansolsson on 20 Sep 2016 07:43 UTC**
Replying to [comment:1 hansolsson]:
> Together with #2004 these are all the cases in MSL.

Well, see also #2066

modelica-trac-importer, 2017-01-15:
**Modified by beutlich on 18 Oct 2016 12:29 UTC**

beutlich, 2017-01-31:
@kristinmajetta @HansOlsson I fixed the undeclared `tableOnFile` bug in branches master and maint/3.2.2. The remaining issue undeclared `useFilter` bug is dealt by #2150. Thus, this issue can be closed.

closed 2017-01-31 by beutlich