modelica/ModelicaStandardLibrary #2006: Names of CubicInterpolation_DP() und CubicInterpolation_MFLOW() are misleading

modelica-trac-importer, 2017-01-15:
**Reported by wischhusen on 6 Jun 2016 07:42 UTC**
Modelica.Fluid.Dissipation.Utilities.Functions.General.CubicInterpolation_DP() calculates a Reynolds number from other Re-inputs but does not take a pressure difference as input
Modelica.Fluid.Dissipation.Utilities.Functions.General.CubicInterpolation_MFLOW() calculates a lambda but does not take a mass flow as input

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/2006

modelica-trac-importer, 2017-01-15:
**Modified by beutlich on 14 Sep 2016 19:06 UTC**

wischhusen, 2018-03-21:
How to auto-convert models that have function calls where the name or path of the function has changed? It seems that convertClass() should somehow have the functionality. But it does not work in my tool.

`equation
Re=Modelica.Fluid.Dissipation.Utilities.Functions.General.CubicInterpolation_MFLOW(1, 2, 3, 4, 5);`

If we change the name of the function how can we convert the call in user libraries?

beutlich, 2018-11-12:
> If we change the name of the function how can we convert the call in user libraries?

@HansOlsson Is it generally possible?

HansOlsson, 2018-11-13:
>>If we change the name of the function how can we convert the call in user libraries?
> @HansOlsson Is it generally possible?

Yes, convertClass should convert function calls as well (the specification just discusses converting classes so it should be covered, and Dymola has implemented it). I don't know if there is a specific issue in this case.

wischhusen, 2018-11-13:
We can give it a try.

PR #2780 (2019-03-07): Renaming of functions CubicInterpolation_DP and CubicInterpolation_MFLOW
close #2006
closed 2019-03-07 by hubertus65