modelica/ModelicaStandardLibrary #2008: Make ModelicaIO.h usable from a C++ environment

modelica-trac-importer, 2017-01-15:
**Reported by ottot on 10 Jun 2016 10:03 UTC**
The functions in ModelicaIO.h should be marked as `extern "C"` when compiled with C++ compiler, see attached patch.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/2008

modelica-trac-importer, 2017-01-15:
**Comment by hansolsson on 13 Jun 2016 11:13 UTC**
We should avoid having defines with generic names such as "EXTERN" since that is likely to cause conflicts. 

Either have no define (requires a bit more) - or use a common prefix, such as the existing "MODELICA_".

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 13 Jun 2016 13:04 UTC**
There already is EXTERN defined in ModelicaMatIO.h.

modelica-trac-importer, 2017-01-15:
**Comment by hansolsson on 13 Jun 2016 13:23 UTC**
Replying to [comment:2 beutlich]:
> There already is EXTERN defined in ModelicaMatIO.h.

Ah, yes, we should thus rename both of them. If necessary with another ticket.
We are lucky that we have NOT stumbled on this yet, but e.g. f2c uses "Extern" and not "EXTERN" (with completely different definition).

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 13 Jun 2016 13:31 UTC**
Right. Will rename to MODELICA_EXTERN (with context of this ticket).

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 23 Jun 2016 20:40 UTC**
Hans: Should we instead utilize the MODELICA_EXPORT macro which alreads is used in ModelicaFFT.c, ModelicaInternal.c, ModelicaRandom.c, ModelicaStrings.c and even in ModelicaIO.c.

It is
```c
#if !defined(MODELICA_EXPORT)
#   define MODELICA_EXPORT
#endif
```

Proposal: Should we
* rename EXTERN to MODELICA_EXPORT in ModelicaMatIO.h,
* add MODELICA_EXPORT in ModelicaStandardTables.h,
* remove MODELICA_EXPORT from ModelicaIO.c and add it to ModelicaIO.h instead?

This way it will be consistent throughout the C Sources.

modelica-trac-importer, 2017-01-15:
**Comment by hansolsson on 28 Jun 2016 14:37 UTC**
Replying to [comment:6 beutlich]:
> @Hans: Should we instead utilize the MODELICA_EXPORT macro which alreads is used in ModelicaFFT.c, ModelicaInternal.c, ModelicaRandom.c, ModelicaStrings.c and even in ModelicaIO.c.
>
> It is
> ```
> #!c [...]

Yes, that seems like a good general solution.

I assume it would be something like:
```c
#if !defined(MODELICA_EXPORT)
#if defined(__cplusplus)
#   define MODELICA_EXPORT extern "C"
#else
#   define MODELICA_EXPORT
#endif
```
I hope that works in all cases, since we have for Dymola noticed that there are a few additional cases for various variants (declaration vs. definition, function vs. data).

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 29 Jun 2016 13:21 UTC**
Replying to [comment:7 hansolsson]:
> I assume it would be something like:
> ```
> #!c
> #if !defined(MODELICA_EXPORT)
> #if defined(__cplusplus)
> #   define MODELICA_EXPORT extern "C"
> #else
> #   define MODELICA_EXPORT
> #endif
> ```

This would only go to the three header files ModelicaIO.h, ModelicaMatIO.h and ModelicaStandardTables.h then, as only these files may get included from a C++ module. Right?

modelica-trac-importer, 2017-01-15:
**Comment by hansolsson on 29 Jun 2016 14:05 UTC**
Replying to [comment:8 beutlich]:
> Replying to [comment:7 hansolsson]:
> This would only go to the three header files ModelicaIO.h, ModelicaMatIO.h and ModelicaStandardTables.h then, as only these files may get included from a C++ [...]

Yes.

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 29 Jun 2016 16:50 UTC**
ModelicaMatIO.h does not contain functions visible for the Modelica environment. Thus I am going to keep the EXTERN there as it is now.

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 30 Jun 2016 06:26 UTC**
Or rename EXTERN to either MATIO_EXTERN or MODELICAMATIO_EXTERN.

beutlich, 2017-04-05:
Resolved by d574148e3a53588da9b8303ee1596c217a2ac74c in master.

closed 2017-04-05 by beutlich