modelica/ModelicaStandardLibrary #2017: Modeling of MSL.Electrical.Analog.OpAmps.SignalGenerator

modelica-trac-importer, 2017-01-15:
**Reported by efredriksson on 21 Jun 2016 07:55 UTC**
I have a question regarding the model SignalGenerator. Running the model in multiple tools result in that we have one state c.v with discontinuous first derivative. The discontinuities wouldn't be a problem if they where associated with events but in this model, there are no event generating expressions. So my question is if this is an appropriate model? 

Attached is a figure of the state and state derivative. Notice the jumps in the derivative and that there are no events in this model. 

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/2017

dietmarw, 2017-01-16:
[image]

beutlich, 2017-03-28:
I believe it could be improved if Modelica.Electrical.Analog.Ideal.IdealizedOpAmpLimted models the limiter the same way as Modelica.Blocks.Nonlinear.Limiter.

beutlich, 2017-10-14:
After 9d5bc998bbe08a153fd622622400e744704a4257 Modelica.Electrical.Analog.Examples.OpAmps.SignalGenerator generates 20 state events for the voltage saturation.

closed 2017-10-14 by beutlich