modelica/ModelicaStandardLibrary #2018: standardized interface between Modelica and PLC

modelica-trac-importer, 2017-01-15:
**Reported by ahaumer on 21 Jun 2016 16:12 UTC**
From industry I heard the wish of "Virtual Putting into Operation". The task would be to create a (standardized) interface between Modelica models of the plant and a PLC (e.g. Siemens S7, B&R, Rockwell, Beckhoff, Moeller, ...). This could be a soft-PLC running on a PC (no need for real time) or a real PLC, the model running in real time on a real-time capable PC. In case of a real PLC of course the user has to provide the physical I/O to the PLC. The advantages are clear: The developer can test his PLC-program long before the plant is available. A company could train people on certain automatisation applications without having demo plants. This reminds me somehow about the FMI-standardization. Any comments?

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/2018

modelica-trac-importer, 2017-01-15:
**Modified by beutlich on 20 Dec 2016 18:32 UTC**

fhgd, 2019-04-16:
I have the same question. Any progress?

chrbertsch, 2019-04-16:
Isn't FMI the solution for this?

See, e.g.,
https://flumes.iei.liu.se/modprod/modprod2017_proceedings/modprod2017-day1-talk02b-Ruediger_Kampfmann_FMI_for_P
https://www.beckhoff.de/TE1420/
https://www.beckhoff.de/TE1420/LC.pdf

beutlich, 2019-04-16:
> 
> 
> I have the same question. Any progress?

AFAIK, there is no development here.

Did you see https://github.com/TechnicalBuildingSystems/ModelicaADS?