modelica/ModelicaStandardLibrary #2031: Modelica.Electrical.Analog.Basic.Inductor, Transformer, etc. render poorly on non-white backgrounds

modelica-trac-importer, 2017-01-15:
**Reported by stefanv on 20 Jul 2016 15:02 UTC**
The annotations for Modelica.Electrical.Analog.Basic.Inductor and Transformer draw their coils by drawing full circles, and then drawing a white rectangle over the portion that is not to be shown. This renders poorly if the background on which the icons are being rendered is not pure white. Attached are two examples.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/2031

modelica-trac-importer, 2017-01-15:
**Comment by stefanv on 20 Jul 2016 15:04 UTC**
Note that the correct way to implement these would be to use the starting and ending angles for the ellipse annotations to only draw the desired half of the circle in the first place.

modelica-trac-importer, 2017-01-15:
**Modified by stefanv on 20 Jul 2016 15:14 UTC**

modelica-trac-importer, 2017-01-15:
**Comment by stefanv on 23 Sep 2016 14:49 UTC**
This ticket cannot be fixed until ticket #2045 is resolved and MCP-0026 is accepted and implemented.

modelica-trac-importer, 2017-01-15:
**Comment by stefanv on 4 Oct 2016 13:08 UTC**
There is an alternative that can fix the issue for these icons immediately, which is to use Lines with smooth=Smooth.Bezier to draw the coils of the inductors. For example, here is an implementation for Modelica.Electrical.Analog.Basic.Inductor:

```mo
  Line(points={{-60,0},{-60,6},{-51,15},{-45,15},{-39,15},{-30,6},{-30,0}},
       color={0,0,255}, smooth=Smooth.Bezier),
  Line(points={{-30,0},{-30,6},{-21,15},{-15,15},{-9,15},{0,6},{0,0}}, 
       color={0,0,255}, smooth=Smooth.Bezier),
  Line(points={{0,0},{0,6},{9,15},{15,15},{21,15},{30,6},{30,0}}, 
       color={0,0,255}, smooth=Smooth.Bezier),
  Line(points={{30,0},{30,6},{39,15},{45,15},{51,15},{60,6},{60,0}}, 
       color={0,0,255}, smooth=Smooth.Bezier),
  ...
```

Here is a list of all the icons I've found that suffer from the problem in this ticket, and that can be fixed by the above technique:

- Modelica.Electrical.Analog.Inductor
- Modelica.Electrical.Analog.SaturatingInductor
- Modelica.Electrical.Analog.Transformer
- Modelica.Electrical.Analog.M_Transformer
- Modelica.Electrical.Ideal.IdealTransformer
- Modelica.Electrical.Ideal.IdealGyrator

modelica-trac-importer, 2017-01-15:
**Comment by majetta on 4 Oct 2016 14:18 UTC**
Fixed for the models mentioned above in ffe4cfd84a163dc4dcf46c31dce96a75dd507e9a and 4761f88e2f0c20d2487aafaf44ed28716be311ee.
Thanks to Stefan Vorkoetter who kindly provided the updated models.

beutlich, 2017-02-01:
@kristinmajetta I squash-cherry-picked your commits from master and pushed them by 0bff3f04e93fd2fd7a39b6f4394e4fb7188ba48f also to maint/3.2.2, such that this issue can be closed.

christiankral, 2017-03-22:
This issue also applies to Modelica.Electrical.Machines.

christiankral, 2017-03-22:
Fixed.

christiankral, 2017-08-17:
There is also an issue with white background and not matching the icon design guidelines of ``Modelica.UsersGuide.Conventions.Icons`` that needs to be fixed

christiankral, 2017-08-18:
There are also still some issues in FundamentalWave

commit 94670c94 (2017-08-17): Fixed #2031 for FluxTubes
commit 1c7135e6 (2017-08-18): Fixed #2031 for FluxTubes
closed 2017-08-18 by christiankral