modelica/ModelicaStandardLibrary #2032: Several text annotations in Magnetic/FluxTubes.mo have a zero width extent

modelica-trac-importer, 2017-01-15:
**Reported by stefanv on 22 Jul 2016 10:36 UTC**
Several instances of annotations for the text "+" and "-" have an extent wherein the left and right bounds for the extent are equal. This will result in these annotations not being displayed (or rather, displayed in a 0-point font).

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/2032

modelica-trac-importer, 2017-01-15:
**Comment by hansolsson on 19 Aug 2016 08:26 UTC**
If only the width is zero then based on #1636 they should be displayed and the text fitted to a larger area.

modelica-trac-importer, 2017-01-15:
**Comment by stefanv on 19 Aug 2016 10:39 UTC**
However, that functionality is not yet in any published specification, so the meaning of the existing annotations with x1=x2 are currently undefined and should not be in the MSL.

beutlich, 2017-02-03:
As stated by @ceraolo in #1636 this issue is not only specific for Magnetic.FluxTubes, but also for e.g., Blocks. I just noticed that also SimulationX does not behave as clarified in upcoming Modelica 3.4.

For fixed strings (e.g, "+" or "-") it should not be a problem to avoid zero-width extents by providing a large enough width. For parametric strings (e.g., "%name") there always is the danger of horizontal truncation. Making the extent ridiculously wide should be possible as workaround.

@HansOlsson How easy can you figure out all occurences of zero-width extents for text strings? Can you even send a PR for the MSL 3.2.2 maintenance branch? Thanks a lot. This probably also should go to the master branch then as long it is not clear if the next major MSL will follow Modelica 3.3, 3.4 or 4.0.

HansOlsson, 2017-02-06:
Finding all occurrences of zero-width extent would take a couple of minutes.  
But automatically generating appropriate sizes to generate a pull-request is more complicated - and to me it would seem like a step backwards, and it would also seem that even when have an MSL following Modelica 3.4 we will not undo this change.

beutlich, 2017-02-06:
Could you please tell a number how many extents are affected?

HansOlsson, 2018-01-15:
> Could you please tell a number how many extents are affected?

88 classes as far as I could tell (using an older version of MSL; so could differ slightly in trunk).
Many of them are for parametric cases, like "f=%f".

beutlich, 2019-10-23:
Since MSL v4.0.0 is based on Modelica language specification version 3.4 the issue is clarified.

closed 2019-10-23 by beutlich