modelica/ModelicaStandardLibrary #2036: Pulse source does not allow for symetrical signal

modelica-trac-importer, 2017-01-15:
**Reported by anonymous on 3 Aug 2016 07:58 UTC**
The current pulse source model allows for asymmetrical pulses : for example, without offset, when the pulse starts it will be between 0 and 1.

It would be good to add an option which allows a pulse with both aspects below
* without offset : before the pulse starts, the value is zero
* with symmetrical values : after it starts, its value is for example between -1 and 1

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/2036

beutlich, 2017-02-12:
The CombiTimeTable block can be used to model symmetrical pulse, sawtooth or trapezoid signals.

beutlich, 2017-08-18:
Closed with proposed workaround.

closed 2017-08-18 by beutlich