modelica/ModelicaStandardLibrary #2047: Electrical Digital Delay docs mention initout

modelica-trac-importer, 2017-01-15:
**Reported by jriel on 17 Aug 2016 18:53 UTC**
The documentation for the InertialDelay and InertialDelaySensitive components mention initout as a parameter.  The actual parameter name is y0.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/2047

modelica-trac-importer, 2017-01-15:
**Comment by clauss on 19 Aug 2016 09:43 UTC**
Thanks for reporting that. The text was changed with f2599a08626b0373449616e779bd15d0ea3384cc.

modelica-trac-importer, 2017-01-15:
**Changelog modified by clauss on 19 Aug 2016 09:43 UTC**
documentstion corrected

modelica-trac-importer, 2017-01-15:
**Comment by dietmarw on 19 Aug 2016 10:08 UTC**
Probably a good idea to fix this in the maintenance branch as well. I saw that you've also changed:

```diff
-  annotation(Dialog(group="table data definition", enable = tableOnFile,
+  annotation(Dialog(group="table data definition", enable = true,
```

in the same commit. That is probably some other bug that was fixed there? Is there already a ticket on that? If not it would probably make sense to create one to document the fix.

modelica-trac-importer, 2017-01-15:
**Modified by sjoelund&#x2e;se on 19 Aug 2016 10:54 UTC**

modelica-trac-importer, 2017-01-15:
**Changelog modified by sjoelund&#x2e;se on 19 Aug 2016 10:54 UTC**
documentation corrected

modelica-trac-importer, 2017-01-15:
**Comment by clauss on 19 Aug 2016 11:26 UTC**
I will fix the documentation error of the inertial delay models in the maintenance branches later on. Therefore, I will keep the ticket open.

The annotation changes (enable = tableOnFile ...) are redone by deb73c75f9faaef22b24276851d8f15af24c1f15. This issue is not quite clear. It refers to #2003, and came in this ticket accidentially.

dietmarw, 2017-01-16:
@ECClauss, @kristinmajetta Has this been done?

kristinmajetta, 2017-01-16:
No, this has not been done yet

--
Kristin Majetta
Wissenschaftliche Mitarbeiterin
Automatisierungs-und Regelungssysteme
Fraunhofer-Institut für Integrierte Schaltungen IIS
Institutsteil Entwicklung Adaptiver Systeme EAS

Zeunerstraße 38, 01069 Dresden
Telefon +49 351 4640-782
mailto:kristin.majetta@eas.iis.fraunhofer.de
www.eas.iis.fraunhofer.de

tbeu, 2017-01-27:
Point to @JoeRiel.

beutlich, 2017-01-27:
Point to @JoeRiel.

beutlich, 2017-01-31:
@kristinmajetta I squash-cherry-picked f2599a08626b0373449616e779bd15d0ea3384cc and deb73c75f9faaef22b24276851d8f15af24c1f15 from master to maint/3.2.2, such that this issue can be closed.

closed 2017-01-31 by beutlich