modelica/ModelicaStandardLibrary #2049: Incorrect doc strings in RotorSaliencyAirGap

modelica-trac-importer, 2017-01-15:
**Reported by jriel on 22 Aug 2016 16:26 UTC**
The doc strings for the magnetic flux variables of the  Modelica.Magnetic.FundamentalWave.BasicMachines.Components.RotorSaliencyAirGap component describe the variables as "Complex magnetic potential difference".  That should be "Complex magnetix flux".

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/2049

modelica-trac-importer, 2017-01-15:
**Comment by jriel on 22 Aug 2016 17:23 UTC**
Also, there is a typo in the description:
"the rotor it is equipped with **to**..."; change **to** to **two**.

modelica-trac-importer, 2017-01-15:
**Comment by jriel on 22 Aug 2016 17:26 UTC**
Further change.  In quoted text, elide the "it".  The phrase should read "the rotor is equipped with ..."

modelica-trac-importer, 2017-01-15:
**Comment by christiankral on 24 Aug 2016 19:07 UTC**
The "to" and "it" also apply to `Modelica.Magnetic.QuasiStatic.FundamentalWave.BasicMachines.Components.RotorSaliencyAirGap`.

Updates are performed in cb6dd1d1d5d30a7ec65b331980125a6373673f23.

modelica-trac-importer, 2017-01-15:
**Modified by christiankral on 24 Aug 2016 19:07 UTC**

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 25 Aug 2016 06:36 UTC**
Please set milestone and release notes fields correctly when closing tickets. Thank you.

modelica-trac-importer, 2017-01-15:
**Modified by christiankral on 25 Aug 2016 15:02 UTC**

modelica-trac-importer, 2017-01-15:
**Changelog modified by christiankral on 25 Aug 2016 15:02 UTC**
Corrected doc strings and documentations of RotorSaliencyAirGap

modelica-trac-importer, 2017-01-15:
**Modified by beutlich on 29 Aug 2016 17:58 UTC**

closed 2017-01-15 by modelica-trac-importer

beutlich, 2017-02-15:
@JoeRiel @christiankral Also cherry-picked to maint/3.2.2 branch.