modelica/ModelicaStandardLibrary #2052: No header for ModelicaRandom.c

modelica-trac-importer, 2017-01-15:
**Reported by efredriksson on 31 Aug 2016 15:04 UTC**
Currently there is no ModelicaRandom.h header file with function declarations (for the functions in ModelicaRandom.c) to use with the ModelicaExternalC library.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/2052

modelica-trac-importer, 2017-01-15:
**Comment by anonymous on 31 Aug 2016 17:37 UTC**
Confirmed. And so do the other files, Strings, FFT, Internal. I guess it is missing for historical reasons where the C files where included. Not sure if header files are really needed.

modelica-trac-importer, 2017-01-15:
**Comment by efredriksson on 1 Sep 2016 08:09 UTC**
Replying to [comment:1 anonymous]:
> Confirmed. And so do the other files, Strings, FFT, Internal. I guess it is missing for historical reasons where the C files where included. Not sure if header files are really needed.

You would only need function declarations for your interface (functions you are going to call from Modelica code), so it is probably not needed for Internal.

It is of course possible to generate these function declarations from your Modelica code. But it is still nice to have the function declarations in a .h file, C compilers will then check that the 2 function declarations are the same.

modelica-trac-importer, 2017-01-15:
**Comment by anonymous on 1 Sep 2016 08:15 UTC**
A Modelica tool cannot both generate the prototype and include a .h file. When using `Include="..."`, it might not be a prototype, but an inline static function instead.

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 2 Sep 2016 11:59 UTC**
Since there is nothing to do, I am closing this ticket.

modelica-trac-importer, 2017-01-15:
**Comment by efredriksson on 5 Sep 2016 13:05 UTC**
Replying to [comment:4 beutlich]:
> Since there is nothing to do, I am closing this ticket.

What is the motivation for that there is nothing to do?

modelica-trac-importer, 2017-01-15:
**Comment by anonymous on 5 Sep 2016 20:34 UTC**
ModelicaExternalC consists of FFT, Internal, Random and Strings - all w/o header.
ModdelicaIO consists of IO - with header.
ModelicaMatIO consists of MatIO - with header.
ModelicaStandardTables consists of StandardTables - with header.

Missing headers are consistenst within ModelicaExternalC itself, but not for all four C libs.

If you ask for Random header, you should also ask for FFT, Internal and Strings header.

modelica-trac-importer, 2017-01-15:
**Comment by efredriksson on 12 Sep 2016 09:34 UTC**
Replying to [comment:6 anonymous]:
> ModelicaExternalC consists of FFT, Internal, Random and Strings - all w/o header.
> ModdelicaIO consists of IO - with header.
> ModelicaMatIO consists of MatIO - with header.
> ModelicaStandardTables [...]

You are correct. All those modules are called externally from Modelica code. I am changing the ticket to reflect this.

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 20 Sep 2016 10:34 UTC**
Hans: What do you think? Should we add ModelicaFFT.h, ModelicaInternal.h, ModelicaRandom.h and ModelicaStrings.h in C-Sources? This would make it consistent throughout all external libraries ModelicaExternalC, ModelicaIO, ModelicaMatIO and ModelicaStandardTables. See also 2008.

modelica-trac-importer, 2017-01-15:
**Comment by hansolsson on 22 Sep 2016 10:13 UTC**
Replying to [comment:9 beutlich]:
> @Hans: What do you think? Should we add ModelicaFFT.h, ModelicaInternal.h, ModelicaRandom.h and ModelicaStrings.h in C-Sources? This would make it consistent throughout all external libraries ModelicaExternalC, ModelicaIO, ModelicaMatIO and ModelicaStandardTables. See also 2008.

On one hand we don't need to add them, the Modelica specification handles that case.

On the other hand, see three slight advantages:
* Consistency
* Clearly marking the interface of the other c-files; so that one can directly see which routines are externally visible.
* The header will be used both for building the library and in the simulation tools, so less risk of inconsistencies.

Obviously those header files (if we add them) should be as minimal as possible, i.e. no unnecessary include etc.

beutlich, 2017-04-05:
Resolved by d574148e3a53588da9b8303ee1596c217a2ac74c in master.

closed 2017-04-05 by beutlich