modelica/ModelicaStandardLibrary #2053: Issues in the M_Oline model

modelica-trac-importer, 2017-01-15:
**Reported by massimo&#x2e;ceraolo on 2 Sep 2016 17:44 UTC**
I think the MSL 3.2.2 Modelica.Electrical.Analog.Lines.M_OLine has several issues, expecially in its documentation.
Please see the enclosed word file.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/2053

dietmarw, 2017-01-16:
[Ticket.docx](https://github.com/modelica/Modelica/files/708136/Ticket.docx)

beutlich, 2017-01-16:
Point to @ceraolo.

beutlich, 2017-08-07:
@kristinmajetta @ECClauss Did you had a look at [Ticket.docx](https://github.com/modelica/Modelica/files/743084/Ticket.docx)?

christophclauss, 2019-08-04:
T'd like to postpone this issue after MSL4.00, since I expect some basic consideration to be necessary which take me too much time now.

beutlich, 2020-02-09:
> 
> 
> T'd like to postpone this issue after MSL4.00, since I expect some basic consideration to be necessary which take me too much time now.

I believe (after fixing the incorrect value of G<sub>13</sub> by 6148f5343550b0707729d010e6a56cba7e67e9cc)  the remaining issues are documentation issues only. E.g., the matrix images for C and G need to be updated, since they contain errors

[image]

max-privato, 2020-02-09:
> I believe (after fixing the incorrect value of G13 by [6148f53](https://github.com/modelica/ModelicaStandardLibrary/commit/6148f5343550b0707729d010e6a56cba7e67e9cc)) the remaining issues are documentation issues only. E.g., the matrix images for C and G need to be updated, since they contain errors
> 
> [image]

In my "Ticket,docx" I suggested to use a multiplier outside the matrices to make them clearer. E.g. 1.e-7 outside L and 1.e-12 outside C. I still believe it would be better.
More important, I highlighted that the C matrix in the model *does not*  contain the C matrix comparing in the vector equation Ia-Ir=(G+jwC)Vr but the components of a network of (positive) capacitances. To make this clear I proposed a deep change in the figure showing capacitances and inductances, with mutual couplings.
A final note. I found several misprints in my document ticket.docx. If deemed useful, I could supply an updated version.

beutlich, 2020-02-09:
> In my "Ticket,docx" I suggested to use a multiplier outside the matrices to make them clearer. E.g. 1.e-7 outside L and 1.e-12 outside C. I still believe it would be better.

See #3398.

dietmarw, 2022-03-15:
@max-privato To me it looks like #3398 took care of this issue or is there anything else left that has not be rectified yet?

max-privato, 2022-03-16:
It is really strange that #3398 changed the way matrices are shown in M_Oline docu:
previously there was already  #3748, in which the M_Oline documentation was fully updated, not only for the matrices but also in many other directions:
- inclusion of mutual couplings between inductors
- elimination of the internal ground pin (substituted with refPin)
- general cleaning

dietmarw, 2022-03-16:
Yes then maybe #3748 is probably the one that really fixed this issue. #3398 was just the change of matrix representation. So I consider this now as fixed.

closed 2022-03-16 by dietmarw