modelica/ModelicaStandardLibrary #2054: Partially erroneous unit of Resistance

modelica-trac-importer, 2017-01-15:
**Reported by massimo&#x2e;ceraolo on 3 Sep 2016 06:42 UTC**
the units of measure of quantities, when expressed in words (not symbols) are all lowercase: ampere, volt, henry, etc.
See for instance https://www.nist.gov/pml/special-publication-330.

Therefore I think that the unit for resistance should be set to "ohm" instead of "Ohm".

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/2054

modelica-trac-importer, 2017-01-15:
**Comment by hansolsson on 5 Sep 2016 13:03 UTC**
I agree that we are not following that standard, but changing to "ohm" would not fix the problem.
\--
To be clear - most units in Modelica are the unit symbols, but "Ω" is non-ASCII and thus an exception is made.

5.1 in the linked publication state that symbols are lowercase - unless derived from a proper name ("m" for meter and "A" for Ampere).
(At some times with an exception for the liter: both "l" and "L" have been ok.)

The unit names are words that as default **start** with lower-case letter (so "degree Celsius", since "Celsius" is a proper name and the "C" isn't at the start!)

However, as any other word the unit names can be capitalized. Ohm is the unit name for resistance, is thus correct - since "Ohm" starts the sentence.
\--
However, the linked standard states that one should not mix unit names and unit symbols, which mean that there is no solution for "Resistivity" - "Ωm" - and we need to handle that case as well.

Another issue is "deg" - since abbreviations are not ok.
\--
An alternative would be to use "Ω" and degree-symbol "°" - since Modelica now handles Unicode.

modelica-trac-importer, 2017-01-15:
**Comment by massimo&#46;ceraolo on 5 Sep 2016 15:19 UTC**
yes, the issue of this, seemingly simple question, is that complex!

Everything you say is totally right.
Especially the last sentence.
I agree: if we are able to manage "Ω" in our MSL (and we should, since we use UTF-8) the best solution is to use this, so that also composed units are correct.

beutlich, 2017-01-16:
Point to @ceraolo.

ceraolo, 2023-12-13:
> Note that even if we have UTF-8, the idea to use them for unit-strings was rejected in MAP-Lang: [modelica/ModelicaSpecification#2142 (comment)](https://github.com/modelica/ModelicaSpecification/issues/2142#issuecomment-435701077)
> 
> Thus I don't see how we can find a good solution for this.
I see.
At this point, I think this ticket can be closed.

beutlich, 2023-12-14:
> At this point, I think this ticket can be closed.

Clsoing as proposed by OP.

closed 2023-12-14 by beutlich