modelica/ModelicaStandardLibrary #2055: Underdetermined initialization of PointGravityWithPointMasses2

modelica-trac-importer, 2017-01-15:
**Reported by henrikt on 6 Sep 2016 10:16 UTC**
The point mass part of `PointGravityWithPointMasses2` has the usual 12 degrees of freedom of a fixed body.  The initialization is given by two point constraints and two velocity constraints, which counts to 12, but the problem is that two degrees of freedom are redundant: the distance between the point is known, and the derivative of that distance is zero.  Therefore, the initialization problem has an infinite number of solutions, making the model not suitable for inclusion in MSL.

To solve the problem of redundant constraints without changing the equation count, the constraints need to involve a third point mass:
```
initial equation
  pointMass1.r_0 = {3, 0, 0};
  pointMass1.v_0 = {0, 0, -1};
  pointMass3.r_0[1] = 2;
  pointMass3.r_0[2] = 1;
  pointMass3.v_0[1] = 0;
  pointMass3.v_0[2] = 0;
  pointMass5.r_0[3] = 1;
  pointMass5.v_0[3] = -1;
```

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/2055

beutlich, 2017-02-03:
@henrikt-ma This underdetermined initialization also also appears for the reference system, right?

henrikt-ma, 2017-02-07:
No, the reference system is initialized as a normal body; position and velocity in a single point, together with an orientation and angular velocity:
```
PointMass pointMass1(r_0(start = {3, 0, 0}, each fixed = true), v_0(start = {0, 0, -1}, each fixed = true), angles_fixed = true, w_0_fixed = true, r_CM = {0, 0, 0})
```

tobolar, 2018-01-10:
Actually, neither the initial conditions provided in the example nor the suggested initial conditions are unique. In both cases the (non fixed) start angles of freeMotion influence the initial orientation.
So it seems to me the only proper solution is to directly initialize those angles, e.g. using `Modelica.Mechanics.MultiBody.Frames.from_nxy`.

HansOlsson, 2018-01-10:
Another alternative seems to be to initialize 4 different positions as follows:
```
  PointMass pointMass1(r_0(start={3,0,0}, each fixed=true), v_0(start={0,0,-1},
        each fixed=true)) ;
  PointMass pointMass3(r_0(start={2,1,0}, fixed={true,false,false}), v_0(start={0,0,-1},
        fixed={true,false,false}));
  PointMass pointMass4(r_0(start={2,-1,0}, fixed={false,false,true}), v_0(start={0,0,-1}, fixed={false,false,true}));
  PointMass pointMass5(r_0(start={0,0,1}, fixed={false,false,true}), v_0(start={0,0,-1}, fixed={false,false,true}));
```

tobolar, 2018-01-11:
@HansOlsson Due to the idea of the example your solution is fine and good understandable to a user. 
It could be also initialized similarly for three points only, as @henrikt-ma suggests:
```
  PointMass pointMass1(
    r_0(start={3,0,0}, each fixed=true),
    v_0(start={0,0,-1}, each fixed=true)) ;
  PointMass pointMass3(
    r_0(start={2,1,0}, fixed={true,true,false}),
    v_0(start={0,0,-1}, fixed={true,true,false}));
  PointMass pointMass5(
    r_0(start={0,0,1}, fixed={false,false,true}),
    v_0(start={0,0,-1}, fixed={false,false,true}));
```
but then the initial positions must be consistent for all three points. In your solution, there is no problem when the user vary either of the coordinate - the initial orientation should always be found when physically meaningful.

commit 1dc20c70 (2018-01-15): MultiBody: Improved initialization of example PointGravityWithPointMasses2

Resolves #2055
closed 2018-01-15 by tobolar