modelica/ModelicaStandardLibrary #2056: Proper definition of Modelica.Constants.{eps,small}

modelica-trac-importer, 2017-01-15:
**Reported by hansolsson on 6 Sep 2016 13:33 UTC**
The values are now given in ModelicaServices - but my problem isn't the values, but the definitions.

In Modelica.Constants it says: 
  final constant Real eps "Biggest number such that 1.0 + eps = 1.0";
  final constant Real small "Smallest number such that small and -small are representable on the machine";
  final constant Real inf "Biggest Real number such that inf and -inf are representable on the machine";
\--
The name "inf" not well-chosen - but apart from that according to the definitions we have on an normal intel-based computer:
inf=1.7976931348623157E+308
eps =1.1102230246251565E-016
small=4.94065645841247E-324 (denormalized number)

Compare this with the C-standard:
#define DBL_EPSILON     2.2204460492503131e-016 
/\* smallest such that 1.0+DBL_EPSILON != 1.0 \*/

#define DBL_MIN         2.2250738585072014e-308 
/\* min positive value \*/

(The actual C11-standard makes it clear that DBL_MIN is a normalized floating point number.)

The previous values of eps and small were conservative estimates such that 1+eps>1 and 1/small was defined. The impact of using eps such that 1+eps=1 or denormalized small is unknown.

I would prefer to map this to DBL_MAX, DBL_EPS and DBL_MIN - and thus adapt the definitions so that this is true.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/2056

modelica-trac-importer, 2017-01-15:
**Comment by hansolsson on 6 Sep 2016 13:40 UTC**
In C11:
DBL_EPS is:
the difference between 1 and the least value greater than 1 that is representable in the given floating point type

DBL_MIN is:
minimum normalized positive floating-point number

DBL_MAX is:
maximum representable finite floating-point number !

AHaumer, 2019-08-05:
see #3081 for further remarks

PR #4042 (2023-06-15): Give a proper definition of machine constants.
They are compatible with text in the C-standard (so one can use DBL_EPS, DBL_MIN and DBL_MAX), and usable in practice.
Closes #2056
commit 879247ef (2023-06-15): Give a proper definition of machine constants.
They are compatible with C, and usable.
Closes #2056