modelica/ModelicaStandardLibrary #2058: Improved version of ideal transmission lines

modelica-trac-importer, 2017-01-15:
**Reported by jriel on 8 Sep 2016 18:13 UTC**
The current implementation of Modelica.Electrical.Analog.Lines.TLine1 (and TLine2 and TLine3) uses four delay elements.  Because delay elements are expensive to instantiate, a better approach is to use only two delay elements, one for the forward propagation, the other for the reverse.  Here is a suggested rewrite:
```
model TLine
    "Lossless transmission line with characteristic impedance Z0 and transmission delay TD"

    // Improved version of Modelica.Electrical.Analog.Lines.TLine1.
    // It uses only two delay elements, rather than four in the original.

    extends Modelica.Electrical.Analog.Interfaces.TwoPort;
    parameter Modelica.SIunits.Resistance Z0(start=1)
        "Characteristic impedance";
    parameter Modelica.SIunits.Time TD(start=1) "Transmission delay";
protected
    // Declare forward and reverse currents and voltages.
    Modelica.SIunits.Current i1f, i1r, i2f, i2r;
    Modelica.SIunits.Voltage v1f, v1r, v2f, v2r;
equation
    assert(Z0 > 0, "Z0 has to be positive");
    assert(TD > 0, "TD has to be positive");
    // continuity
    v1 = +v1f + v1r;
    v2 = +v2f + v2r;
    i1 = +i1f + i1r;
    i2 = -i2f - i2r;
    // delay
    v1r = delay(v2r, TD);
    v2f = delay(v1f, TD);
    // characteristic impedance
    v1f = +Z0*i1f;
    v2f = +Z0*i2f;
    v1r = -Z0*i1r;
    v2r = -Z0*i2r;
end TLine;
```

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/2058

modelica-trac-importer, 2017-01-15:
**Comment by rfranke on 9 Sep 2016 06:04 UTC**
It appears questionable why the model uses `delay` at all. This introduces hidden states without initial conditions in the model. The vertical bar in `T` stands for a line (shunt) capacitance that is not present in the model. A `delay` is used instead. Does this unphysical line model make sense?

The `PowerSystems.AC1ph_DC` components are similar to `Modelica.Electrical.Analog` (1-phase time domain models with two voltages and two currents in each connector). `PowerSystems.AC1ph_DC.Lines.Tline` models `ne` line segments with `ne+1` series impedances and `ne` shunt capacitances in between.
```mo
model Tline "T transmission line, 1-phase"
  extends Ports.Port_p_n;
  extends Partials.LineBase;

  PS.Voltage[2,ne] v(start = transpose(fill(v_start, ne)));
  PS.Current[2,ne1] i(start = transpose(fill(i_start, ne1)));
protected 
  final parameter Integer ne1=ne + 1;

initial equation 
  if dynType == Types.Dynamics.SteadyInitial then
    der(v) = zeros(2,ne);
    der(i) = zeros(2,ne1);
  elseif dynType == Types.Dynamics.FixedInitial then
    v = transpose(fill(v_start, ne));
    i = transpose(fill(i_start, ne1));
  end if;

equation 
  i[:, 1] = term_p.i;
  i[:, ne1] = -term_n.i;

  C*der(v) + G*v = i[:, 1:ne] - i[:, 2:ne1];
  L*der(i) + diagonal(R)*i = [[2*(term_p.v - v[:, 1])], v[:, 1:ne - 1] - v[:, 2:ne], [2*(v[:, ne] - term_n.v)]];
end Tline;
```

See: https://github.com/modelica/PowerSystems/blob/master/PowerSystems/AC1ph_DC/Lines.mo

If you want to change the `Modelica.Analog.Line` models, then you might consider removing `delay` completely and introduce a physical formulation instead. Do also note the more popular PI-Line with capacitances at the terminals and line impedance in between.

modelica-trac-importer, 2017-01-15:
**Comment by massimo&#46;ceraolo on 10 Sep 2016 22:44 UTC**
@rfranke, why you say that transmission delay is not physical?

In a sense it is much more physical than PI-line model and even multi-pi line model.
Consider for instance a 300 km line. If you apply a step to its first end in a real line that step reaches the other end after 1 ms, and this can be simulated by TLine1. If a lumped network is used, instead, part of the step reaches (unrealistically) the receiving end immediately after the application of the step. I.e. the signal moves ad an infinite speed!
I have and example that I use for my students in which I compare OLine and TLine and show that when the number of OLine's PIs raises the results becomes nearer to the TLine one, without fully reaching it. 
You say that the shunt C is not present in the model. Indeed the delay is the result of the presence of distributed L and distributed C, as in any real-life line.
So, for me, having both TLine and OLine is very good.
Naturally, if the same result is obtained with more efficient code, it would be very welcome.

modelica-trac-importer, 2017-01-15:
**Comment by rfranke on 11 Sep 2016 17:14 UTC**
I say the `TLine1` model is not physical because it is behavioural. You know that electrons travel with speed of light at most, i.e. 1ms for 300km. `TLine1` is based on a transport delay resulting from the characteristics of the analytical solution. The timeout is just the beginning of the step response. The whole response is an order of magnitute slower and it is described quite well with PI or T line elements.

`Modelica.Analog.Electrical.Lines.OLine` in fact is a physical T-line model. It exhibits balance equations and the behavior results from their solution. Having no `delay` with hidden states, you can invest into a number of line segments with real states, in order to better approximate a step response.

Modelica is great in plugging physical models together for obtaining the system response. Using the physical `OLine`, you can for instance leave one end of the 300km line unconnected -- then you get the step response of a long open line. `TLine1` gives a structural singular model when leaving one end unconnected.

Moreover, the physical `OLine` model covers losses as well -- losses of long lines are typically more important than small delays at the beginning of a step response.

Anyway, of course `TLine1` might be useful for some people and the formulation proposed by jriel  above is of course smarter.

modelica-trac-importer, 2017-01-15:
**Comment by rfranke on 12 Sep 2016 08:04 UTC**
It appears that there is neither an example nor a test for `Modelica.Electrical.Lines`.

Could something appropriate be added along with the proposed change?

modelica-trac-importer, 2017-01-15:
**Comment by clauss on 12 Sep 2016 12:16 UTC**
Yes, I will provide examples. But I connot do it immediately, sorry.

modelica-trac-importer, 2017-01-15:
**Comment by massimo&#x2e;ceraolo on 13 Sep 2016 06:15 UTC**
Replying to [comment:4 rfranke]:
> It appears that there is neither an example nor a test for `Modelica.Electrical.Lines`.
> 
> Could something appropriate be added along with the proposed change?

I will also upload a few examples just as ideas for possible inclusion in MSL. I will first make them neater.

modelica-trac-importer, 2017-01-15:
**Comment by massimo&#x2e;ceraolo on 13 Sep 2016 06:52 UTC**
Well. I want to contribute fast.
I attach a small package lines2058.mo and lines2058.pptx showing how OLine and TLine models can, together, be useful in travelling waves ans electromagnetics transients studies.

modelica-trac-importer, 2017-01-15:
**Comment by rfranke on 13 Sep 2016 15:20 UTC**
@massimo: Thank you for the nice example!

modelica-trac-importer, 2017-01-15:
**Comment by massimo&#46;ceraolo on 13 Sep 2016 19:04 UTC**
glad that you like!
I can add  few more words. Generally speaking I would say that Modelica is a tool for simulating lumped parameters systems. These systems are governed by ordinary differential 
Distributed parameter systems, instead,  are governed by partial differential equations, and experience finite delays: one important system of these is exactly constituted by the transmission lines.
In this sense TLine is a very special modelica model: instead of being based on partial differential equations, it implements the known result for a lossless line, i.e. the transmission delay.
Though so simplified, this model has in some respects more correspondence to the physics of the line than an R-C network, as my examples show. Not in all respects, however.
That's why it is good for me to have both models.
Maybe some examples, e.g. based on ideas behind my models, can make this clearer to final users.

modelica-trac-importer, 2017-01-15:
**Comment by jriel on 13 Sep 2016 20:17 UTC**
Here's a refactorization of my improved version.  It eliminates the current variables.
```
model TLine
    "Lossless transmission line with characteristic impedance Z0 and transmission delay TD"

    // Improved version of Modelica.Electrical.Analog.Lines.TLine1.
    // It uses only two delay elements, rather than the four in the original.

    extends Modelica.Electrical.Analog.Interfaces.TwoPort;
    parameter Modelica.SIunits.Resistance Z0(start=1)
        "Characteristic impedance";
    parameter Modelica.SIunits.Time TD(start=1) "Transmission delay";
protected
    // Declare forward and reverse voltages.
    Modelica.SIunits.Voltage v1f, v1r, v2f, v2r;
equation
    assert(Z0 > 0, "Z0 has to be positive");
    assert(TD > 0, "TD has to be positive");
    v1 = +v1f + v1r;
    v2 = +v2f + v2r;
    v1r = delay(v2r, TD);
    v2f = delay(v1f, TD);
    Z0*i1 = +v1f - v1r;
    Z0*i2 = -v2f + v2r;
end TLine;
```

modelica-trac-importer, 2017-01-15:
**Comment by jriel on 13 Sep 2016 20:37 UTC**
One more refactorization.  The auxiliary voltages can also be eliminated and the entire model replaced with essentially one Modelica statement.  That there is a single delay function shouldn't improve the efficiency as it is vectorized.
```
model TLine
    "Lossless transmission line with characteristic impedance Z0 and transmission delay TD"
    extends Modelica.Electrical.Analog.Interfaces.TwoPort;
    parameter Modelica.SIunits.Resistance Z0(start=1)
        "Characteristic impedance";
    parameter Modelica.SIunits.Time TD(start=1) "Transmission delay";
equation
    assert(Z0 > 0, "Z0 has to be positive");
    assert(TD > 0, "TD has to be positive");
    {v1-Z0*i1,v2-Z0*i2} = delay({v2+Z0*i2,v1*Z0*i1},TD);
end TLine;
```

beutlich, 2017-01-16:
Point to @ceraolo.

ceraolo, 2017-02-03:
Well, I did not check the maths, but tried the model and seems to not to work properly. Naturally I've first corrected v1*z0*i1 into v1+z0*i1.

christophclauss, 2019-08-04:
A lot of tests are necessary to solve this issue, I'd like postpone it after MSL4.0.0 to get enough time.

dietmarw, 2019-08-04:
@christophclauss Could you be more specific as to what "more" time means. I mean there is still at least 6 month of development time for 4.0.0. Dismissing this already now makes it sound like a bad excuse to not deal with the issue at all. This also applies for the others. So could you please evaluate again if you really think that there is no time to do this within about a  6 month window?