modelica/ModelicaStandardLibrary #2060: Modelica.Fluid.Examples.Tanks.TanksWithOverflow does not respect variable bounds

modelica-trac-importer, 2017-01-15:
**Reported by agnes&#x2e;ramle on 12 Sep 2016 11:24 UTC**
The model Modelica.Fluid.Examples.Tanks.TanksWithOverflow does only initialize if the max and min bounds are ignored and the initial values are outside of these bounds. 
An example of a variable with this problem is `overflow.port_a.p`.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/2060

modelica-trac-importer, 2017-01-15:
**Modified by beutlich on 15 Sep 2016 06:28 UTC**

chria, 2018-05-21:
We could also update the example, see the following patch (sorry for it not being a pull request)
[decreasesHeightTanksWithOverflow.txt](https://github.com/modelica/ModelicaStandardLibrary/files/2021873/decreasesHeightTanksWithOverflow.txt)

PR #2539 (2018-05-28): Decrease height in Modelica.Fluid.Examples.Tanks.TanksWithOverflow
As proposed by @chria.
Close #2060.
closed 2018-05-28 by casella