modelica/ModelicaStandardLibrary #2063: Update start value in Modelica.Fluid.Examples.InverseParameterization

modelica-trac-importer, 2017-01-15:
**Reported by agnes&#x2e;ramle on 16 Sep 2016 12:48 UTC**
Would it be possible to update pump.m_flow_start to not start at zero flow. The current 0 flow causes pump.port_b.m_flow to be on the wrong branch in the actualStream operator in
```
 pump.Hb_flow = pump.port_a.m_flow * actualStream(pump.port_a.h_outflow) + pump.port_b.m_flow * actualStream(pump.port_b.h_outflow);
```

The suggestion is to set pump.m_flow_start=0.5.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/2063

modelica-trac-importer, 2017-01-15:
**Comment by hansolsson on 19 Sep 2016 12:48 UTC**
Changing the defaults in such generic models does not seem like a good solution, and might change existing results.

However, I can see a numeric issue with using the definition of actualStream at that point - in particular that we select inconsistent branches for the two actualStream.

chria, 2018-05-21:
Would it be possible to update the start value in the example instead? See the attachment (sorry for it not being a pull request)
[updateStartInverseParameterization.txt](https://github.com/modelica/ModelicaStandardLibrary/files/2021902/updateStartInverseParameterization.txt)

beutlich, 2018-05-22:
@chria Can you please check if 62ae89374d43ca90b390d9561f0d9f149ef265db already fixes the issue on your side.

chria, 2018-05-23:
@beutlich I've checked and it fixes the issue, thanks!

beutlich, 2018-05-23:
Thanks for confirmation.

Resolved by 62ae89374d43ca90b390d9561f0d9f149ef265db in master.

closed 2018-05-23 by beutlich