modelica/ModelicaStandardLibrary #2064: Add new block RealFFT with examples

modelica-trac-importer, 2017-01-15:
**Reported by ahaumer on 16 Sep 2016 18:41 UTC**
This is the missing ticket for the new feature RealFFT added by 71fb9c898168aff6c11a658c57d40bfb2e3b0382, da9a10a0a4ba52c7cf1c2f0c92e2150ba00a9f37, 26c93cf66d8d9f047d686d33e87918dbb4fbd08b and 5c138dcf97895edf7c9a7b1d17bc16b4b6e6771f.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/2064

beutlich, 2017-01-31:
Done.

beutlich, 2017-02-27:
The icon of Modelica.Blocks.Math.RealFFT looks rather naked. Please improve!

AHaumer, 2018-05-18:
added icon

beutlich, 2018-05-18:
> added icon

It seems to me you forgot to push. If you add an "close #2064" or "fix #2064" to your commit message, the issue will close autmaotically once you push.

commit 7850cb98 (2018-05-20): resolves #2064
closed 2018-05-20 by AHaumer