modelica/ModelicaStandardLibrary #2068: Vector.interpolate doc should mention it is linear

modelica-trac-importer, 2017-01-15:
**Reported by pierre&#x2e;haessig on 22 Sep 2016 14:35 UTC**
I believe that the docstring of Vector.interpolate doesn't mention the fact it's linear. Looking at the source, it is!

Here is a minimalistic proposition: https://github.com/modelica/Modelica/pull/6

First sentence becomes:

The function call Vectors.interpolate(x,y,xi) interpolates **linearly** in vectors ....

(I didn't put bold in the actual code, although an italic formatting may be appropriate)

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/2068

modelica-trac-importer, 2017-01-15:
**Modified by beutlich on 30 Sep 2016 10:37 UTC**

beutlich, 2017-01-31:
@pierre-haessig Resolved in branches master and maint/3.2.2.

closed 2017-01-31 by beutlich