modelica/ModelicaStandardLibrary #2071: Integrator Block leads to wrong results

modelica-trac-importer, 2017-01-15:
**Reported by mmans on 27 Sep 2016 10:33 UTC**
In Dymola (2016, 2016FD01, 2017) (32,64bit), as well as OpenModelica 1.9.1 (r22929), the use of an Integrator Block leads to wrong results. 

In our case we are using the integrator block to calculate the simulated energy of a model with a power output. We think, that this should work correct, but it does not (with a stock simulation tolerance of 1e-4). Only with a tolerance of 1e-10 we get nearly right results. If you need further information, a model, or screenshots we are looking forward to provide these.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/2071

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 27 Sep 2016 13:54 UTC**
Providing a simple test model that demonstrates the expected and current behavior is really of help.

modelica-trac-importer, 2017-01-15:
**Comment by mmans on 27 Sep 2016 14:34 UTC**
I added a simple model with a CombiTimeTable, reading a timeseries from a txt file and an Integrator Block. If i simulate this with a tolerance of 1e-4 it calculates wrong values, if i decrease the tolerance to 1e-10 i'll get better results.

modelica-trac-importer, 2017-01-15:
**Comment by otter on 27 Sep 2016 14:39 UTC**
Replying to [comment:2 mmans]:
> I added a simple model with a CombiTimeTable, reading a timeseries from a txt file and an Integrator Block. If i simulate this with a tolerance of 1e-4 it calculates wrong values, if i decrease the tolerance to 1e-10 i'll get better results.

Please, explain the expected result and why you think the simulation result is wrong (note, most integration methods introduce "artifical" damping and therefore it is well-known that a system "looses" energy during numerical integration, even if the energy is conserved for the exact solution).

modelica-trac-importer, 2017-01-15:
**Comment by mmans on 12 Oct 2016 08:47 UTC**
Replying to [comment:3 otter]:
> Replying to [comment:2 mmans]:
>
> Please, explain the expected result and why you think the simulation result is wrong (note, most integration methods introduce "artifical" damping and therefore it is [...]

Sorry for the late reply. We had a look into it again and we weren't aware of the "artificial" damping fact of integrators.
What we expected as a simulation result by using an integrator block was, the area below our input curve. With the use of a combitimetable and a linear interpolation method between the given points of the table, we couldn't use the integrator block to actualy get the excact result of the area below the curve, read from the table.
Assuming that this way of using an integrator will introduce artificial damping to our models, what are the "workarounds" for this? Is the only method to achieve accurate results by reducing the tolerance? How does the integrator handle the integration? Does it get further signals which are not thrown with a greater tolerance?

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 12 Oct 2016 11:36 UTC**
Discussion sounds similar to the reopened case from #1627. You can check if TimeTable block works as expected provided the table data is given as array parameter.

beutlich, 2017-01-29:
@PMehrfeld Closed as duplicate of #2080.

closed 2017-01-29 by beutlich