modelica/ModelicaStandardLibrary #2072: Illegal use of delay() in Modelica.Electrical.Digital.Delay.TransportDelay

modelica-trac-importer, 2017-01-15:
**Reported by stefanv on 28 Sep 2016 21:11 UTC**
The model `Modelica.Electrical.Digital.Delay.TransportDelay` contains the following:

```mo
x_delayed = LogicValues[integer(delay(Integer(pre(x)),delayTime))];
```

Both `x` and `x_delayed` are of type `Modelica.Electrical.Digital.Interfaces.Logic`, which is an enumeration type.

This code, specifically `delay(Integer(...),...)`, is illegal. Section 3.7.2 of the specification states that all the arguments to `delay` must be a subtype of `Real`. The `Integer(...)` function returns an `Integer`, and section 6.1 of the specification states that `Integer` is<ins>not</ins> a subtype of `Real`.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/2072

modelica-trac-importer, 2017-01-15:
**Comment by hansolsson on 3 Oct 2016 07:52 UTC**
I believe that we should improve the specification to handle this, since the goal here is to delay a discrete integer value and preserve its discreteness.

For normal delay signals it is often necessary to interpolate between two real values, (which isn't that complicated), but interpolation between two LogicValues is neither straightforward nor desired.

Specifically the delayed signal should trigger an event when it changes, and just changing it to be real with some casts will not give the desired result.

modelica-trac-importer, 2017-01-15:
**Comment by stefanv on 4 Oct 2016 14:09 UTC**
_The following comment is from our chief simulation engine developer:_

The proposed approach may cause some difficulty in practial integration of a system to any appreciable multiple of the delay.

Consider a model with a discrete delay of 1 sec. and a sample that occurs every 0.099 sec., and further assume that the discrete variable that is computed from the delay is also involved in the delay and affected by the sample.

In the range 0<t<1 there will be events at times:
0.099, 0.198, 0.297, ...

In the range 1<t<2 there will be events at times:
1.089, 1.188, 1.287, ...
but also induced events (from the delay) at:
1.099, 1.198, 1.297, ...

In the range 2<t<3 this becomes:
2.079, 2.178, 2.277, ...
with induced events
2.089, 2.188, 2.287, ...
2.099, 2.198, 2.297, ...

So overall we will have:
10 events in 0<t<1
20 events in 1<t<2
30 events in 2<t<3
40 events in 3<t<4
50 events in 4<t<5
...

It is clear that this problem rapidly becomes quite difficult to integrate, getting slower and slower over time, as more events accumulate.

For continuous signals involved in ODE this is generally not a problem. For example, for y'(t)=y(t-1) a jump dincontinuity at t=0 is damped to a cusp at t=1 which becomes a change in the 2nd order derivative at t=2, etc. Basically the discontiuities drop in order over time, which avoids the necessity of retaining them across the delay for all time.

Perhaps there is a different way to model this component that does not require a delay acting on a discrete variable?

modelica-trac-importer, 2017-01-15:
**Comment by hansolsson on 4 Oct 2016 14:42 UTC**
Replying to [comment:2 stefanv]:
> _The following comment is from our chief simulation engine developer:_
>
> The proposed approach may cause some difficulty in practial integration of a system to any appreciable multiple of the delay. [...]

I don't think we can fully do that.

However:
As far as I understand transport delay and inertial delay are the standard components for handling delays in circuits with somewhat different characteristics.

So, the first solution is to use the inertial delay instead - which avoids the problem, since at most one change is delayed.
We might make it clearer that inertial delay is the recommended model.

The second attempt at an solution would be to have a clocked transport delay (using Modelica 3.3 synchronous or sampled) - but to me that should be part of a new clocked Digital library, and even then it would just make the problem obvious, not solve it.

In normal circuits the delay would be an integer multiple of the sample-time, and then we avoid the event-explosion, but here the common sampling rate is 1ms and we then at the start do something every 1000 and 99th sample - but after awhile we will do something every sample in that case.

kristinmajetta, 2017-07-05:
As long as delay() cannot have arguments of type enumeration, we (Christoph and myself) introduced an auxiliary variable xr of type Real:
 `xr = Integer(pre(x));`
xr is protected so that it can be removed anytime without breaking the backward compatibility

SHA-1: d7252ec27b0ea2f2f986bcc52d43b4ce03064bb5

tbeu, 2017-07-06:
This now assigns an integer type to the real variable. Is this legal?

HansOlsson, 2017-07-06:
> This now assigns an integer type to the real variable. Is this legal?

Yes, an integer value can be assigned to a real variable, and for integer _literal expressions_ it is quite common as well - both in the specification and in the. standard library.

tbeu, 2017-07-07:
Thanks for clarification. Thus thos issue can be closed then?

HansOlsson, 2017-07-10:
Yes, unless I missed something. If we want to extend the delay-operator to enumeration variables we can add that later with a new ticket.

closed 2017-10-04 by kristinmajetta