modelica/ModelicaStandardLibrary #2073: Document Visual Studio linker optimization options for ModelicaExternalC and ModelicaStandardTables

modelica-trac-importer, 2017-01-15:
**Reported by beutlich on 29 Sep 2016 18:44 UTC**
Peter Lundberg contacted me that his application depending on ModelicaInternal.c hangs or freezes after updating to MSL 3.2.2.

With #1433 (for ModelicaExternalC) and #1550 (for ModelicaStandardTables) the header gconstructor.h from glib project was introduced to automatically initialize/terminate the mutex/lock as module constructor/destructor required for thread-safety.

As stated in comment 5 of #1550 the module constructor/destructor is removed by the default linker optimization options of the Release configuration of Visual Studio. And calling EnterCriticalSection on a uninitialized critical section (i.e., InitializeCriticalSection was not called before) can lead to undefinded behavior.

The Visual Studio linker optimization options for ModelicaExternalC and ModelicaStandardTables should be clearly documented.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/2073

modelica-trac-importer, 2017-01-15:
**Modified by beutlich on 29 Sep 2016 18:46 UTC**

beutlich, 2017-01-29:
Resolved by 11e95b66328c7ec1ad02819411a8c746b47f880c.

beutlich, 2017-03-24:
The documentation Needs to be revised since the removal of the module ctor/dtor also depends on the /GL option (whole program optimization).

beutlich, 2017-03-29:
Related to https://bugzilla.gnome.org/show_bug.cgi?id=752837

beutlich, 2017-04-05:
@d94pn Improved in master.

closed 2017-04-05 by beutlich