modelica/ModelicaStandardLibrary #2076: Invalid SI unit VA

modelica-trac-importer, 2017-01-15:
**Reported by sjoelund&#x2e;se on 4 Oct 2016 11:56 UTC**
SIUnits contains the following declaration:
```mo
  type ApparentPower = Real (final quantity="Power", final unit="VA");
```

But the specification says:
> Modelica enforces the ISO recommendation that each multiplication operator is explicitly written out in formal specifications. For example, Modelica does not support "Nm" for newtonmeter, but requires it to written as "N.m".

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/2076

modelica-trac-importer, 2017-01-15:
**Comment by rfranke on 4 Oct 2016 16:30 UTC**
Wikipedia says that volt-ampere (VA) is a unit of its own, see:
  https://en.wikipedia.org/wiki/Volt-ampere

modelica-trac-importer, 2017-01-15:
**Comment by sjoelund&#46;se on 4 Oct 2016 17:10 UTC**
The specification is very vague on which derived units in the SI system should be supported though. I checked quite a few webpages like http://physics.nist.gov/cuu/Units/units.html and none of them list VA as a derived unit. I guess they all consider it implicit V\*A, which Modelica does not support. Note that the given URL also lists J (N\*m) as a derived unit (which is explicitly forbidden in the spec), but fails to mention VA as a separate unit, which is why I think it should be listed as `V.A` in SIunits so tools know it is Volt-Ampere and not a custom unit `VA`.

modelica-trac-importer, 2017-01-15:
**Comment by rfranke on 5 Oct 2016 05:27 UTC**
Modelica tools anyway need to know "var" that stands for voltage-ampere reactive and is also equivalent to V\*A -- quantity Power.
```mo
  type ReactivePower = Real (final quantity="Power", final unit="var");
```

Do we make tools for users or users for tools?

modelica-trac-importer, 2017-01-15:
**Comment by hansolsson on 5 Oct 2016 15:41 UTC**
Replying to [comment:3 rfranke]:
> Modelica tools anyway need to know "var" that stands for voltage-ampere reactive and is also equivalent to V\*A -- quantity Power.
> ```mo
>   type ReactivePower = Real (final quantity="Power", final unit="var");
> ```
> 
> Do we make tools for users or users for tools?

Hopefully tools for users.

One pragmatic solution would be to explicitly list the ones we allow and their definitions (or possibly remove some of them).

For just the unit-part we have (all found in MSL) - in addition to some allowed non-SI units that can be used with SI (min, h, d, l, bar, B, eV - and spelling ° as deg):
http://physics.nist.gov/cuu/Units/outside.html

Power:
  1VA=1V.A Replace by "V.A"?
  1var=1V.A
Rotational speed:
  1rpm=2\*pi/60 rad/s
  "r" as in 1r/min=1rpm
  "rev" as in 1rev/min=1rpm Replace by "r"?
Temperatures:
  1degRk=5/9 K
  1degF=5/9 K with offset
Electrical dipole moment (Modelica.Media.Interfaces.Types.DipoleMoment):   
  1debye=(1/299792458)\*1e-21 C·m
Sound (as far as I can see these are not in the SI-system):
  phon 
  sone

modelica-trac-importer, 2017-01-15:
**Comment by rfranke on 5 Oct 2016 18:12 UTC**
Replying to [comment:4 hansolsson]: 
> One pragmatic solution would be to explicitly list the ones we allow and their definitions (or possibly remove some of them).
>
> For just the unit-part we have (all found in MSL) - in addition to some [...]

The link also lists `t` for metric ton -- a unit that is missed a lot when dealing with power plants. Many measure produced steam in `t/h`. It's a pity that Modelica does not support it.

The solution might indeed be to explicitly list all supported Non-SI units. I hope that anyone agrees that e.g. `deg` and `h` must be supported. Could we also add `t`?

Electrical engineers would be happy to find `VA` and `var` supported. (People who think this is not needed might ask themselves to explain the difference between apparant power and reactive power. ;-))

ceraolo, 2017-02-22:
Official units of measure for electromagnetism are found in IEC 80000-6.
From there I read:
Item 6-57, quantity name:  "apparent power", unit name: "volt ampere", symbol V.A
Item 6-59, quantity name:  "complex power",  unit name: "volt ampere", symbol V.A
Item 6-60, quantity name:  "reactive power", two units allowed: 
     6-60a: unit name "volt ampere ", symbol V.A
     6-60b  unit name "var", symbol var.  Note: 1 var := 1 V.A

AHaumer, 2017-12-03:
Please do NOT replace var by V.A or VA - it is important to distinguish apparent power (V.A), reactive power (var - note: in lower case!) and active power (W).

dietmarw, 2019-06-20:
So is the conclusion we leave it as is and allow "var" and "VA" etc?

christiankral, 2019-06-20:
I guess we shall leave "var" and change "VA" to "V.A" as this is in line with the convention of specifying units.

beutlich, 2019-06-20:
Can you create a prull request for it then? Thanks.

PR #2982 (2019-06-21): Change unit of apparent power from "VA" to "V.A"
close #2076
closed 2019-06-21 by beutlich