modelica/ModelicaStandardLibrary #2078: Modelica.Math.Vectors.interpolate lacks derivative

modelica-trac-importer, 2017-01-15:
**Reported by hansolsson on 10 Oct 2016 13:14 UTC**
The simplest solution would be:
```
 annotation(smoothOrder (normallyConstant=x, normallyConstant=y)=100);
```
Another option would be explicitly provide the derivative, which could then give a zero 2nd derivative (since it interpolates linearly as stated in another ticket).

(Note that this will not work in current versions of Dymola.)

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/2078

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 4 Jan 2017 12:44 UTC**
I suppose it is problematic that Modelica.Math.Vectors.interpolate returns two outputs where one is Boolean (i.e., non Real).

modelica-trac-importer, 2017-01-15:
**Comment by hansolsson on 4 Jan 2017 12:48 UTC**
Replying to [comment:1 beutlich]:
> I suppose it is problematic that Modelica.Math.Vectors.interpolate returns two outputs where one is Boolean (i.e., non Real).

Not a problem if providing the derivative-function since "12.7.1 Using the Derivative Annotation" state that outputs not containing reals are skipped:
"The outputs are constructed by starting with an empty list and then in order appending one derivative for each output containing reals."
(The recent MCP for derivatives was records mixing real and non-real).

We might clarify that the same happens if we use smoothOrder.

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 4 Jan 2017 12:55 UTC**
Replying to [comment:2 hansolsson]:
> Replying to [comment:1 beutlich]:
> > I suppose it is problematic that Modelica.Math.Vectors.interpolate returns two outputs where one is Boolean (i.e., non Real).
> 
> Not a problem if providing the derivative-function

So why does [T2078.mo](https://gist.github.com/beutlich/1bf026dc61e8a1fa8e1708d9e8090208) fail to translate?

modelica-trac-importer, 2017-01-15:
**Comment by hansolsson on 4 Jan 2017 13:09 UTC**
Replying to [comment:3 beutlich]:
> Replying to [comment:2 hansolsson]:
> So why does [T2078.mo](https://gist.github.com/beutlich/1bf026dc61e8a1fa8e1708d9e8090208) fail to translate?

Because we currently fail to separate the differentiation of the function from the differentiation of the statement.

If the line is split into two lines it works:
```
  yi = interpolate(x1,y,10*time);
  (,iNew)=interpolate(x1,y,10*time);
```
I meant that from a language perspective it is not problematic; and I will see how to improve the implementation.

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 4 Jan 2017 13:33 UTC**
I am thinking of proposing these cases to the ModelicaCompliance suite.

HansOlsson, 2017-06-16:
Note that Dymola 2018 was updated to handled cases like T2078.mo.

PR #2264 (2018-01-08): specifySmoothness of interpolate function
Resolve #2078 by specifying smoothness as specified in ticket.
closed 2018-01-08 by MartinOtter