modelica/ModelicaStandardLibrary #2079: Parameter enableNoise depends per default on settings in GlobalSeed

modelica-trac-importer, 2017-01-15:
**Reported by tobolar on 13 Oct 2016 08:59 UTC**
Since the availibility of multiple noise signals generally applies for a complete model where they are used, the parameter `enableNoise` in Modelica.Blocks.Interfaces.PartialNoise should be per default dependent on `globalSeed.enableNoise` from Modelica.Blocks.Noise.GlobalSeed

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/2079

modelica-trac-importer, 2017-01-15:
**Modified by beutlich on 13 Oct 2016 20:40 UTC**

beutlich, 2017-02-07:
@tobolar Can you please provide a PR for this enhancement. Thank you very much.

tobolar, 2017-02-08:
Sorry, but what does PR stand for? [http://www.abbreviations.com/PR](http://www.abbreviations.com/PR)

xie-dongping, 2017-02-08:
@tobolar

PR stands for pull request in GitHub.
https://help.github.com/articles/about-pull-requests/

tobolar, 2017-02-08:
I see. Then I will prepare PR.

beutlich, 2017-02-08:
Resolved both in master and maint/3.2.2.

closed 2017-02-08 by beutlich