modelica/ModelicaStandardLibrary #2084: Duplicated functions getMemory and address in Modelica.Electrical.Digital.Memories

modelica-trac-importer, 2017-01-15:
**Reported by beutlich on 18 Oct 2016 12:33 UTC**
There is no need to have have the functions getMemory and address both in Memories.DLATRAM and Memories.DLATRAM.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/2084

modelica-trac-importer, 2017-01-15:
**Comment by hansolsson on 26 Oct 2016 09:45 UTC**
Replying to [ticket:2084 beutlich]:
> There is no need to have have the functions getMemory and address both in Memories.DLATRAM and Memories.DLATRAM.

I assume you mean "There is no need to have have the function getMemory both in Memories.DLATRAM and Memories.DLATROM."

As far as I can see address is not currently used in DLATROM (but could be).

I agree that moving getMemory (and possibly address) could be moved up one level.

modelica-trac-importer, 2017-01-15:
**Comment by anonymous on 26 Oct 2016 09:57 UTC**
You are right, I mixed it up. There is only [DLATRAM.address](https://github.com/svn2github/Modelica_StandardLibrary/blob/master/Modelica/Electrical/Digital.mo#L7445-L7464), but DLATROM uses the same [algorithm](https://github.com/svn2github/Modelica_StandardLibrary/blob/master/Modelica/Electrical/Digital.mo#L7700-L7710) without a separate function, thus kind of inlined.

beutlich, 2017-01-31:
Moving public functions (in or across packages) requires a conversion script, so reschedule Milestone.

beutlich, 2017-04-19:
Resolved in master by preserving backward compatibility.

closed 2017-04-19 by beutlich