modelica/ModelicaStandardLibrary #2086: Line color of Modelica.Electrical.Spice3.Sources.V_constant icon

modelica-trac-importer, 2017-01-15:
**Reported by beutlich on 18 Oct 2016 13:42 UTC**
Should Sources.V_constant be better drawn in blue (instead black) to better match the other components or the connect color?

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/2086

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 18 Oct 2016 13:50 UTC**
Please do also look at Semiconductors.C_Capacitor.

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 18 Oct 2016 13:56 UTC**
And also in Semiconductors.D_DIODE.

kristinmajetta, 2017-02-23:
Changed linecolor from black to blue in Modelica.Electrical.Spice3.Sources.V_constant
73ee16e7ae5408a7eb65123bacec88b2dd2f7e2d

beutlich, 2017-02-27:
Thanks. What about D_DIODE and C_Capacitor?

beutlich, 2017-03-13:
Resolved in master.

closed 2017-03-13 by beutlich