modelica/ModelicaStandardLibrary #2089: Missing comments for result variables in Modelica.Electrical.Spice3

modelica-trac-importer, 2017-01-15:
**Reported by beutlich on 18 Oct 2016 14:15 UTC**
Most of the variables in Modelica.Electrical.Spice3 have no comment

* Semiconductors.M_PMOS
* Semiconductors.M_NMOS
* Semiconductors.M_NMOS2
* Semiconductors.M_PMOS2
* Semiconductors.Q_NPNBJT
* Semiconductors.Q_PNPBJT
* Semiconductors.J_PJFJFET
* Semiconductors.J_NJFJFET
* Semiconductors.D_DIODE
* Internal.Rsemiconductor.ResistorModelLineParams.m_dRshIsGiven
* Internal.Rsemiconductor.ResistorVariables

List is not complete. Since I checked this manually, I stopped here and will resume checking once this ticket is dealt with.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/2089

kristinmajetta, 2017-04-04:
We already discussed this issue years ago. The SPICE3 models were developed from the original SPICE3f source code where none of the variables had explanations.  When transforming the original SPICE3f models to Modelica, we interpreted many of the variables and added comment stringt so them. Unfortunatelly it is not possible to interpret all the variables since not all of them do even have a physical meaning. The models mentioned in this ticket all extend from models of the package Modelica.Electrical.Spice3.Internal which should not be accessed by the user anywhy. This is we we agreed that it is ok to leave some of the variables uncommented.

beutlich, 2017-04-04:
I understand your trouble. It simply does not look mature when checking the model results.
[image]

kristinmajetta, 2018-05-03:
How to proceed here? Should we close that ticket?

dietmarw, 2018-05-03:
I was trying to figure out the source of the original SPICE3 models but fail to find them. The linked user's manual does not seem to have the models themselves. Knowing were exactly the reference comes from/looking at it might help figuring the meaning of the variables. 

I fear if it is not done now and the "link" to the original Spice models is lost then it will be lost forever and hence the trust in these models.

kristinmajetta, 2018-05-09:
The original SPICE3 sources are the base of the Modelica Spice3 models. Originally the Berkeley SPICE sources were written in Fordran, later on converted into C (by Berkeley). Both sources do not contain unit descriptions. About 15 years ago the C soucres were isolated and transformed into a separat C++ model library within a doctoral thesis at Fraunhofer EAS (Leitner: Entwicklung simulatorunabhängiger Modelle für  Halbleiter-Bauelemente mit objekt-orientierten Methoden, ISBN: 978-3-933342-80-5). Within the C++ library the models were rewritten as object oriented models. Out of the C++ library we transformed the models to Modelica - still without units. Unfortunatelly the C++ sources are commercial ones and can't ne distributed. You could possibly try to get the original Fordran and C SPICE sources from Berekley. They should be freely available.

The right way to get the units as well as the description strings is to analyse the equations. Since this is a very complex task due to the huge number of equations, we pospone the ticket to the next milestone.

Christoph and Kristin

christophclauss, 2019-08-04:
This issue need  digging into the SPICE sources, therefore I'd like to postpone it beyond MSL4.0.0