modelica/ModelicaStandardLibrary #2090: Calling Modelica.Math functions in functions of Modelica.Mechanica.MultiBody.Frames

modelica-trac-importer, 2017-01-15:
**Reported by beutlich on 18 Oct 2016 22:12 UTC**
Some functions (like Frames.axisRotation) call the builtin functions (for e.g., sin, cos, atan2 etc.), some other functions (like Frames.planarRotation) call the corresponding functions from Modelica.Math (e.g., Modelica.Math.sin).

* What is the recommended way to call functions with builtin representation in functions?
* What is the reason to call the Modelica.Math functions.
* Why is it not consistent through the library?

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/2090

modelica-trac-importer, 2017-01-15:
**Comment by hansolsson on 26 Oct 2016 09:51 UTC**
To me calling "sin" instead of "Modelica.Math.sin" is sort of a compatibility hack (which was added due to the C-function "sin" - which was later re-interpreted as calling functions in Modelica.Math - unrelated to the C-function).

Thus I prefer the style of using "Modelica.Math.sin".

modelica-trac-importer, 2017-01-15:
**Comment by anonymous on 26 Oct 2016 10:04 UTC**
Replying to [comment:1 hansolsson]:
> Thus I prefer the style of using "Modelica.Math.sin".

Thanks for the explanation. If we call it a _standard_ library I'd like to see it consistent throught the library which it currently is not. And, having "Modelica.Math.sin" instead of simple C "sin" bloats up the library a lot (and may also have an impact on translation performance.)

beutlich, 2017-02-15:
@MartinOtter @HansOlsson Do we agree on calling "Modelica.Math.sin" over "sin"? If yes, I can work on this iusse.

thorade, 2017-02-15:
This seems to be related:
Using `sinh()` works, but using `asinh()` directly fails in Dymola 2017 FD01 (or was it a warning?).
I switched to using the full path everywhere, like `Modelica.Math.asinh()`

sjoelund, 2017-02-15:
@thorade That should be due to `asinh()` not being part of the specification. Its implementation in MSL is: `y := Modelica.Math.log(u + sqrt(u*u + 1));`, so using the full path (or `import Modelica.Math;` ... `Math.asinh()`, etc) is the only solution in that case.

HansOlsson, 2017-02-15:
@beutlich I prefer `Modelica.Math.sin` over `sin` - but some unknown person disagreed citing code bloat and performance. (There is no performance difference in Dymola at least.)

@thorade The functions that are supposed to work are listed in the Modelica specification 3.7.1.2 Built-in Mathematical functions and includes `sinh`, `asin` - but not `asinh` (as @sjoelund  also noted).

beutlich, 2017-02-15:
@HansOlsson If you refer to 
> Comment by anonymous on 26 Oct 2016 10:04 UTC

above, that was written me. I still prefer to have it consistent. The "bloated" was related to Hilding's Modia paper, where he argued against long names. And if there is no performance drop, we have the evidence to make it consitent again.

HansOlsson, 2017-02-15:
> @HansOlsson If you refer to Comment by anonymous on 26 Oct 2016 10:04 UTC above, that was written me.

Ok, and I agree that it is best to have it consistent.

dietmarw, 2017-02-15:
For consistency sake, would it then not be the easiest to have library wide imports in the form of:
```mo
import sin=Modelica.Math.sin
import asin=Modelica.Math.asin
import sinh=Modelica.Math.sinh
import asinh=Modelica.Math.asinh
import cos=Modelica.Math.cos
...
```

dietmarw, 2017-02-15:
Ha even the GH lexer tells you that `asinh` is not a build-in...nice

HansOlsson, 2017-02-15:
> For consistency sake, would it then not be the easiest to have library wide imports in the form of:
> 
> import sin=Modelica.Math.sin

I would prefer not to have that. 
The reasons are that:

- You have to remember all the imports in the library if you just look at the source code.
- People writing new models outside of MSL will be confused/mindlessly copy this boilerplate etc.
- Historically imports were used in that way, and it led to a number of duplicated imports.

dietmarw, 2017-02-15:
So @HansOlsson what you are saying is not using any imports at all in the MSL because it is seen as the reference? Historically I see it more a problem that there were no proper guidelines for the MSL developers, hence each sub-library did duplicate the imports (even within their own sub-library). Having one dedicated  and easy to find section for imports right at the beginning of the top-level package might help improve this situation.

HansOlsson, 2017-02-15:
> So @HansOlsson what you are saying is not using any imports at all in the MSL because it is seen as the reference?  Historically I see it more a problem that there were no proper guidelines for the [...]

I would more say that we should try to avoid imports that are far removed from the use. 
Using an import in a model isn't a problem (even if I believe we even had duplicated imports even in the same model).
Package-wide imports are more problematic - in my opinion.

Obviously tools can help - both for full names (code completion, abbreviated view for presentation) and abbreviated (showing the relevant imports), and in both cases providing links to the function.

beutlich, 2017-02-20:
I fixed the ones in Modelica.Mechanics and Modelica.Blocks.Continuous. I did not touch the other packages.

closed 2017-02-20 by beutlich