modelica/ModelicaStandardLibrary #2091: Utilize outerProduct in functions of Modelica.Mechanics.MultiBody.Frames

modelica-trac-importer, 2017-01-15:
**Reported by beutlich on 19 Oct 2016 06:16 UTC**
Rewrite the term `[e]*transpose([e])` in Frames.planarRotation, Frames.smallRotation, Frames.from_Q, Frames.multipleResolve1, Frames.multipleResolve2, Frames.to_T and Frames.to_T_inv by `outerProduct(e,e)` to avoid the duplicated array construction and make it more readable and economic.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/2091

modelica-trac-importer, 2017-01-15:
**Comment by hansolsson on 26 Oct 2016 10:36 UTC**
Replying to [ticket:2091 beutlich]:
> Rewrite the term `[e]*transpose([e])` in Frames.planarRotation, Frames.smallRotation, Frames.from_Q, Frames.multipleResolve1, Frames.multipleResolve2, Frames.to_T and Frames.to_T_inv by `outerProduct(e,e)` to avoid the duplicated array construction and make it more readable and economic.

I agree with the change.
However, I prefer if it isn't too rushed - since it could possibly cause failure to translate with current version of our tools in some odd scenario. (Will be corrected.)

modelica-trac-importer, 2017-01-15:
**Comment by anonymous on 26 Oct 2016 11:42 UTC**
Replying to [comment:1 hansolsson]:
> Replying to [ticket:2091 beutlich]:
>
> I agree with the change.
> However, I prefer if it isn't too rushed - since it could possibly cause failure to translate with current version of our tools in some [...]

Do not worry, there is no rush.

beutlich, 2017-08-05:
Resolved by dcc02ba3e6229415f3d87ee8522768fd4342513f in master.

closed 2017-08-05 by beutlich