modelica/ModelicaStandardLibrary #2095: Update zlib to latest develop

modelica-trac-importer, 2017-01-15:
**Reported by beutlich on 25 Oct 2016 10:21 UTC**
zlib 1.2.8 is from April 2013 and was not updated since then. There have been som issues fixed in the official develop branch of https://github.com/madler/zlib that are important to have. Therefore I'd like to update zlib in C-Sources to current develop branch.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/2095

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 2 Jan 2017 08:04 UTC**
Resolved by ec046c6ae6cbb897c461090bc53a7c0c0f49c5f8.

modelica-trac-importer, 2017-01-15:
**Changelog modified by beutlich on 2 Jan 2017 08:04 UTC**
Updated zlib to v.1.2.9

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 3 Jan 2017 11:20 UTC**
Updated to v1.2.10 by 801c129cc3800a1e6c27de35eafa59b82fa28932.

modelica-trac-importer, 2017-01-15:
**Changelog modified by beutlich on 3 Jan 2017 11:20 UTC**
Updated zlib to v.1.2.10

beutlich, 2017-01-17:
Update zlib to v.1.2.11

closed 2017-01-27 by beutlich