modelica/ModelicaStandardLibrary #2097: Fix uthash_fatal

modelica-trac-importer, 2017-01-15:
**Reported by beutlich on 26 Oct 2016 08:21 UTC**
The definition of uthash_fatal in ModelicaStandardTables and ModelicaInternal turned out not be suitable. See https://github.com/troydhanson/uthash/issues/92#issuecomment-256253862 for the uthash discussion.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/2097

modelica-trac-importer, 2017-01-15:
**Comment by arthur&#46;j&#x2e;odwyer on 27 Oct 2016 21:20 UTC**
FYI, it might be cleaner to remove the big /\*comment\*/ about the one place uthash_fatal is used, and instead do something like this:

```
#include "uthash.h"
#undef uthash_fatal  // ensure that nowhere in this file uses uthash_fatal by accident
[...]

static double* readTable(const char* tableName, const char* fileName,
                         size_t* nRow, size_t* nCol, int verbose, int force) {
#define uthash_fatal(msg) do { MUTEX_UNLOCK(); ModelicaFormatError("Error: %s\n", msg); return; } while (0)
    double* table = NULL;
    if (tableName != NULL && fileName != NULL && nRow != NULL && nCol != NULL) {
       [...]
    }
    return table;
#undef uthash_fatal
}
```

I don't know if your organization's coding style guidelines would smile on this idiom (#define/#undef anywhere other than the top of the file), but I've found it to be a useful idiom in C codebases that \*do\* permit such a thing.

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 28 Oct 2016 06:22 UTC**
Thank you, Arthur. I really appreciate your external expertise here. I will adopt the locally define/undef style which is cleaner indeed.

In what condition is the `head` if `HASH_ADD_KEYPTR(hh,head,keyptr,keylen_in,add)` fails with uthash_fatal? Is it save to do some clean-up and free `keyptr` and `add` passed to in macro `HASH_ADD_KEYPTR`, like I did in attachment:T2097_2.patch?

modelica-trac-importer, 2017-01-15:
**Comment by sjoelund&#x2e;se on 3 Nov 2016 10:51 UTC**
Caught in the spam filter, arthur.j.odwyer@gmail.com says:

> I'd strongly prefer if you credited yourself rather than me in the changelog, since it's your patch and I wouldn't want anyone blaming (or more importantly, contacting) me with any Modelica issues/questions.
> 
> I suspect you should MUTEX_UNLOCK() \*before\* free()ing those pointers, in order to keep the critical section as small as possible.
> 
> The actual code/semantics of the patch look to me "as good as it's going to get", unless it's acceptable to you to just call `abort()` instead, in which case I now definitely think `abort()` would be safer.
> 
> > In what condition is the `head` if `HASH_ADD_KEYPTR(hh,head,keyptr,keylen_in,add)` fails with uthash_fatal?
> 
> I plan to take a little bit of time this weekend to see whether tableShare would really be left in an "invariant-satisfying" state after an invocation of uthash_fatal; I'll update https://github.com/troydhanson/uthash/issues/93 when I have something. I suspect the answer will be "yuck, generally `head` is in a bad state and you shouldn't touch it again even to HASH_CLEAR it"; but it's certainly safe to `free(keyptr)` and so on as long as you \*don't\* touch `head` again. If you \*do\* touch `head` again... well, watch  https://github.com/troydhanson/uthash/issues/93 for those details.

beutlich, 2017-02-26:
Resolved in master.

closed 2017-02-26 by beutlich