modelica/ModelicaStandardLibrary #2099: Fix documentation of Modelica.Utilities.Streams.error

modelica-trac-importer, 2017-01-15:
**Reported by beutlich on 7 Nov 2016 21:42 UTC**
The documentation of Modelica.Utilities.Streams.error says: "Print the string \"string\" as error message and cancel all actions.". But since Modelica.Utilities.Streams.error calls ModelicaError which behaves as an assert with (default = ) error assertion level, an solver can try recover from the problem. This is demonstrated by the example model T, where models M1 and M2 behave identically.

```mo
model T
  function f
    input Real t_;
    output Real d;
  external "C" d = _f(t_) annotation(
    Include = "
#include \"ModelicaUtilities.h\"
double _f(double t_) {
  static int s = 1;
  if (s && t_ > 0.5) {
    s = 0;
    return -1.;
  }
  return 1.;
}");
  end f;
  model M
    Real x(start=0, fixed=true);
    Real y;
    equation
      y = f(time);
      der(x) = y;
  end M;
  model M1 "M w/ assertion"
    extends M;
    algorithm
      assert(noEvent(y >= 0), "Assert");
  end M1;
  model M2 "M w/ ModelicaError"
    extends M;
    algorithm
      if noEvent(y < 0) then
        Modelica.Utilities.Streams.error("ModelicaError");
      end if;
  end M2;
  M2 m annotation(Placement(transformation(extent={{-100,80},{-80,100}})));
  annotation(uses(Modelica(version="3.2.1")));
end T;
```
Thus, it neither is clear what "cancel all actions" means, nor is it correct, if interpreted as "stop simulation with an error".

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/2099

modelica-trac-importer, 2017-01-15:
**Modified by beutlich on 8 Nov 2016 19:23 UTC**

beutlich, 2017-10-15:
It seems the semantics of Modelica.Utilities.Streams.error is erroneously interpreted as unrecoverable fatal error (which is not available in Modelica, see [mo:#2152](https://trac.modelica.org/Modelica/ticket/2152)).

PR #2367 (2018-01-08): Fix the documentation of Modelica.Utilities.Streams.error
Close #2099.
closed 2018-01-08 by MartinOtter