modelica/ModelicaStandardLibrary #2100: Missing obsolete annotation

modelica-trac-importer, 2017-01-15:
**Reported by hansolsson on 8 Nov 2016 13:45 UTC**
Modelica 3.2 revision 2 added (see #877):
  annotation(obsolete = "message");

The intention is to explain why a particular class is obsolete, and thus this cannot be in the base-class Modelica.Icons.ObsoleteModel

However, it should be classes extending from it - with appropriate message - but that seems to be lacking from all models using it:
Modelica.Electrical.Spice3.Internal.BJT (and sub-classes)
Modelica.Electrical.Spice3.Internal.Bjt3
Modelica.Electrical.Spice3.Internal.Bjt.Bjt3Variables
Modelica.Electrical.Spice3.Internal.Bjt.BjtModelLineVariables
Modelica.Electrical.Spice3.Internal.Bjt.BjtVariables
Modelica.SIunits.Conversions.ConversionIcon
Modelica.Electrical.Spice3.Internal.Mos2.drainCur
Modelica.Electrical.Spice3.Internal.SpiceRoot.initJunctionVoltages
Modelica.Electrical.Spice3.Internal.Functions.junction2SPICE3MOSFET
Modelica.Electrical.Spice3.Internal.Functions.junctionCap
Modelica.Electrical.Spice3.Internal.SpiceRoot.limitJunctionVoltage
Modelica.Electrical.Spice3.Internal.ModelcardBJT
Modelica.Electrical.Spice3.Internal.Mos.mos2CalcCalcTempDependencies
Modelica.Electrical.Spice3.Internal.Mos.mos2CalcInitEquations
Modelica.Electrical.Spice3.Internal.Mos.mos2CalcNoBypassCode
Modelica.Electrical.Spice3.Internal.Mos2.mos2ModelLineParamsInitEquations
Modelica.Electrical.Spice3.Internal.Mos2.mos2RenameParameters
Modelica.Mechanics.Rotational.Interfaces.PartialElementaryOneFlangeAndSupport
Modelica.Mechanics.Translational.Interfaces.PartialElementaryOneFlangeAndSupport
Modelica.Mechanics.Rotational.Interfaces.PartialElementaryTwoFlangesAndSupport
Modelica.Mechanics.Translational.Interfaces.PartialElementaryTwoFlangesAndSupport
Modelica.Magnetic.FundamentalWave.BasicMachines.Components.SaliencyCageWinding_obsolete
Modelica.Magnetic.FundamentalWave.BasicMachines.Components.SymmetricMultiPhaseCageWinding_obsolete
Modelica.Math.tempInterpol1
Modelica.Math.tempInterpol2

Most of them already explain this, but some in text, some in description etc. Using the annotation just ensures consistency.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/2100

christiankral, 2017-03-05:
Added obsolete annotation for maint/3.2.2 in b185b43 and master in 830e52b.

beutlich, 2017-09-06:
Also:
```
Modelica.Mechanics.MultiBody.Sensors.TansformAbsoluteVector
Modelica.Mechanics.MultiBody.Sensors.TansformRelativeVector
Modelica.Icons.TypeComplex
Modelica.SIunits.LoundnessLevel
Modelica.SIunits.Loundness
Modelica.Blocks.Interfaces.Adaptors.*
```

beutlich, 2017-09-06:
Also:
```
Modelica.Icons.Library
Modelica.Icons.Library2
Modelica.Icons.GearIcon
Modelica.Icons.MotorIcon
Modelica.Icons.Info
Modelica.Blocks.Interfaces.BlockIcon
Modelica.Blocks.Interfaces.BooleanBlockIcon
Modelica.Blocks.Interfaces.DiscreteBlockIcon
Modelica.Blocks.Interfaces.IntegerBlockIcon
Modelica.Blocks.Interfaces.partialBooleanBlockIcon
Modelica.Electrical.Digital.Converters.LogicToXO1
Modelica.Electrical.Digital.Converters.LogicToXO1Z
Modelica.Fluid.Icons.VariantLibrary
Modelica.Fluid.Icons.BaseClassLibrary
Modelica.Math.baseIcon1
Modelica.Math.baseIcon2
```

closed 2018-04-29 by beutlich