modelica/ModelicaStandardLibrary #2101: Dimensionally wrong equation in simplified expression for homotopy initialization of pump models

modelica-trac-importer, 2017-01-15:
**Reported by fcasella on 10 Nov 2016 01:08 UTC**
Willi Gerbig reports:

 After I had run one of our models that ran without any errors and warnings before, I got two warnings concerning incompatible units. I have looked at the warnings in more detail then and found out that they semm to refer to two equations that appear in Modelica.Fluid.Machines.BaseClasses.PartialPump.
To be preciser the equations staring with "head = ..." in the lines 142 and 147 seem to cause these warnings. Both equations contain the expression "flowCharacteristic(0)+delta_head_init\*V_flow_single", which is indeed an addition where the units of the two summands are not compatible since "flowCharacteristic(0)" yields "m" as unit and "delta_head_init\*V_flow_single" yields "m^4/s" as unit.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/2101

modelica-trac-importer, 2017-01-15:
**Comment by fcasella on 10 Nov 2016 01:26 UTC**
The right expression is of course 
  flowCharacteristic(0)+delta_head_init\*V_flow_single/V_flow_single_init

Fixed in trunk dc4f6d3e2a0f27b72f7146e72b9cdaaddc08d3a4, 3.2.1 maintenance e903713c2e895ae10d1ebb8d00913609506d1d7d, 3.2.2 maintenance e392ee8df98b6a33b06367d8b919d2f9426437b3

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 11 Nov 2016 06:51 UTC**
Since it is not sure if the next MSL is 3.2.2+build.4, 3.2.3, 3.3.0 or 4.0.0 I updated the milestone.

modelica-trac-importer, 2017-01-15:
**Comment by hansolsson on 11 Nov 2016 14:58 UTC**
Replying to [comment:1 fcasella]:
> The right expression is of course 
>   flowCharacteristic(0)+delta_head_init\*V_flow_single/V_flow_single_init

This ticket is a duplicate of #1995, opened a six months ago and without any response.

The ticket #1995 has the above as one possibility - but states that it for the default values causes division by zero, since V_flow_single_init is as default 0 (based on system.m_flow_start).

modelica-trac-importer, 2017-01-15:
**Comment by fcasella on 11 Nov 2016 15:33 UTC**
In fact, when I was thinking of a quick fix for this problem I thought about the division by zero issue, but then I saw the division by V_flow_single_init showing up in other (dimensionally correct) parts of the code and thought it could be ok.

I'm sorry I missed #1995, I'll fix that ASAP.

I'm going to close this ticket because it's a duplicate of #1995, and also because the issue that it raises (dimensionally wrong equations) has indeed been fixed.

closed 2017-01-15 by modelica-trac-importer

beutlich, 2018-05-07:
Set milestone to never (since it is a duplicate of #1995).

commit dc4f6d3e (2016-11-10): Fixes #2101 in development trunk

git-svn-id: https://svn.modelica.org/projects/Modelica/trunk@9521 7ce873d0-865f-4ce7-a662-4bb36ea78beb
commit e903713c (2016-11-10): Fixes #2101 in 3.2.1 maintenance branch

git-svn-id: https://svn.modelica.org/projects/Modelica/branches/maintenance/3.2.1@9522 7ce873d0-865f-4ce7-a662-4bb36ea78beb
commit e392ee8d (2016-11-10): Fixes #2101 in 3.2.2 maintenance branch

git-svn-id: https://svn.modelica.org/projects/Modelica/branches/maintenance/3.2.2@9523 7ce873d0-865f-4ce7-a662-4bb36ea78beb