modelica/ModelicaStandardLibrary #2102: Modelica.Blocks.Sources.Exponentials should be a block

modelica-trac-importer, 2017-01-15:
**Reported by jriel on 10 Nov 2016 18:32 UTC**
It has type model rather than block.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/2102

modelica-trac-importer, 2017-01-15:
**Comment by jriel on 10 Nov 2016 18:40 UTC**
Similarly for Modelica.Blocks.Routing.BooleanPassThrough, IntegerPassThrough, and RealPassThrough.

modelica-trac-importer, 2017-01-15:
**Modified by beutlich on 11 Nov 2016 07:46 UTC**

beutlich, 2017-02-07:
That was changed in ee9a1a29568c2815a0e568719f25482a50d388c7 from block to model. @DagBruck Did you change the type by accident or purpose?

beutlich, 2017-02-07:
Pointing to @JoeRiel.

beutlich, 2017-02-14:
@DagBruck Any objections to change Exponentials back to block?

beutlich, 2017-02-22:
Resolved in master.

beutlich, 2017-02-22:
Leave a comment

closed 2017-02-22 by beutlich