modelica/ModelicaStandardLibrary #2105: Improve support of MATLAB structure variables

modelica-trac-importer, 2017-01-15:
**Reported by beutlich on 11 Nov 2016 07:16 UTC**
#1840 added the feature to read variables from MATLAB structure variables to MSL 3.2.2. However the feature currently is limitated to 1x1 structures. It would be easy to generalize the feature for arbitrary two-dimensional strcutures by giving giving the structure element, e.g., `a.b` could be generalized to `a(1,1).b` where `a` is a structure and `b` is a real array. Then `a(2,3).b(4,5).c` could also be valid inputs.

If this feature is not considered, then at least the current error message for mxn structures should be improved to point to the dimensional problem and not to the type problem.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/2105

tbeu, 2017-02-01:
@Claytex Thanks for the nice blog at http://www.claytex.com/blog/reading-structures-from-mat-files. Do you have any feature requests?

beutlich, 2017-04-05:
@HansOlsson @sjoelund I kindly ask for your input since I have no idea who else cares and whom I could ask. Thanks for your feedback. I appreciate it.

Options are:
1. Do not add support for multidimensional structs, but improve error message.
2. Add support for multidimensional structs and use MATLAB syntax for array subscription
3. Add support for multidimensional structs and use Modelica syntax for array subscription
4. Use both MATLAB and Modelica syntax 

Which one do you prefer?

beutlich, 2017-04-12:
@Claytex @HansOlsson @MartinOtter Any feedback? Otherwise I will simply close as wontfix.

HansOlsson, 2017-04-12:
I have no specific opinion at this time

beutlich, 2017-04-12:
OK, no need to work on it (now), if there is no public interest.

closed 2017-04-12 by beutlich